MichaIng/DietPi #886: GDBus.Error RPi 3

radimsejk, 2017-04-17:
Here is the Reference code for my issue: 236f2941-36a9-4e5d-8352-748394a43144-0
When the LXDE is started (automatically after boot) this message is shown:

> GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown:
> The name org.freedesktop.PolicyKit1 was not provided by any .service files

What should I do?

Fourdee, 2017-04-18:
@radimsejk 

Thanks for the report.

Can you try installing policy kit to see if the error is disabled?
```
apt-get install policykit-1
reboot
```

radimsejk, 2017-04-18:
This is output in the terminal when I try to install it (and it is the same with the -f flag):
`root@PiBlet:~# apt-get install policykit-1
Reading package lists... Done
Building dependency tree
Reading state information... Done
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 policykit-1 : Depends: libpolkit-backend-1-0 (= 0.105-15~deb8u2) but it is not going to be installed
 teamviewer-host : Depends: libqt5gui5 (>= 5.2.1) but it is not going to be installed
                   Depends: libqt5widgets5 (>= 5.2.1) but it is not going to be installed
                   Depends: libqt5qml5 (>= 5.2.1) but it is not going to be installed
                   Depends: libqt5quick5 (>= 5.2.1) but it is not going to be installed
                   Depends: libqt5webkit5 (>= 5.1.1) but it is not going to be installed
                   Depends: libqt5x11extras5 (>= 5.2.1) but it is not going to be installed
                   Depends: qtdeclarative5-qtquick2-plugin (>= 5.2.1) but it is not installable or
                            qml-module-qtquick2 (>= 5.2.1) but it is not going to be installed
                   Depends: qtdeclarative5-controls-plugin (>= 5.2.1) but it is not installable or
                            qml-module-qtquick-controls (>= 5.2.1) but it is not going to be installed
                   Depends: qtdeclarative5-dialogs-plugin (>= 5.2.1) but it is not installable or
                            qml-module-qtquick-dialogs (>= 5.2.1) but it is not going to be installed
                   Depends: qtdeclarative5-window-plugin (>= 5.2.1) but it is not installable or
                            qml-module-qtquick-window2 (>= 5.2.1) but it is not going to be installed
                   Depends: qtdeclarative5-quicklayouts-plugin (>= 5.2.1) but it is not installable or
                            qml-module-qtquick-layouts (>= 5.2.1) but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).`

Fourdee, 2017-04-18:
@radimsejk 
Thanks,

try:
```
apt-get install -f
```

radimsejk, 2017-04-18:
Yeah, it worked. Thank you :)

Fourdee, 2017-04-18:
Reopened for testing/investigation and possible fix in ```dietpi-software```

Fourdee, 2017-04-19:
Unable to replicate on a fresh LXDE installation Marking as closed.

closed 2017-04-19 by Fourdee