MichaIng/DietPi #887: Using btrfs filesystem

taner1, 2017-04-17:
Is it possible to use btrfs for system partition? I dont see any distro using it but it does have compression for getting more free space and implies wear level algorithm that extends lifespan of sd cards. Also for some benchmarks its faster than obsolote fat32 filesystem.

Invictaz, 2017-04-17:
Is it better than the F2FS used in modern Androids now? Is it backwards compatible so other computers an see it?

taner1, 2017-04-18:
I dont know if its better than f2fs because there is no comparison between two but f2fs has no transparent compression support and btrfs is more mature than it for inquiry of compatibility. Since dietpi focused on lower footprint i think btrfs is more suitable on this purpose.

WolfganP, 2017-04-18:
There's an interesting discussion in https://github.com/raspberrypi/linux/issues/1761 , it seems that implementing btrfs in raspbian and derivatives requires high maintenance for now.

Fourdee, 2017-05-06:
Marking as closed, please reopen if required.

closed 2017-05-06 by Fourdee

taner1, 2017-05-16:
I managed to migrate my rootfs ext4 to btrfs with adding btrfs to initramfs. The btrfs module already in kernel so no need to recompile. With lzo compression there is slight delay at startup time but after booting seems everything is good. It decreased my occupied space 650MB to 420MB. Anyone who using 1-2GB sd  cards i strongly recommend to use btrfs for it . Now finally i can install ffmpeg to my rpi zero :)

trajano, 2017-09-04:
@taner1 how does it fare when you have unexpected power failures?  Does it recover well?