MichaIng/DietPi #889: DietPi-Software | Are MongoDB Server

phonicmouse, 2017-04-21:


phonicmouse, 2017-06-27:
What about this? @Fourdee 
😄

Fourdee, 2017-06-28:
https://docs.mongodb.com/manual/tutorial/install-mongodb-on-debian/
```
apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv 0C49F3730359A14518585931BC711F9BA15703C6
echo -e "deb http://repo.mongodb.org/apt/debian jessie/mongodb-org/3.4 main" > /etc/apt/sources.list.d/mongodb-org-3.4.list
apt-get update
apt-get install -y mongodb-org
```
- 🈴 Lacks ARMv7 ARMv8 binaries
```
W: Failed to fetch http://repo.mongodb.org/apt/debian/dists/jessie/mongodb-org/3.4/Release  Unable to find expected entry 'main/binary-arm64/Packages' in Release file (Wrong sources.list entry or malformed file)

W: Failed to fetch http://repo.mongodb.org/apt/debian/dists/jessie/mongodb-org/3.4/Release  Unable to find expected entry 'main/binary-armhf/Packages' in Release file (Wrong sources.list entry or malformed file)

```
- x86_32 x86_64 only

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

abhinavgupta, 2018-01-08:
Hi, installing mongodb is still an issue in aarch64 machines for Dietpi. I am trying to install mongodb on Pine64 version of Dietpi and getting similar errors:

W: Failed to fetch http://repo.mongodb.org/apt/debian/dists/jessie/mongodb-org/3.4/Release  Unable to find expected entry 'main/binary-arm64/Packages' in Release file (Wrong sources.list entry or malformed file)

W: Failed to fetch http://repo.mongodb.org/apt/debian/dists/jessie/mongodb-org/3.4/Release  Unable to find expected entry 'main/binary-armhf/Packages' in Release file (Wrong sources.list entry or malformed file)

Fourdee, 2018-01-08:
@abhinavgupta 

The repo links you listed, only provide x86 packages. They lack ARM.
```
Origin: mongodb
Label: mongodb
Suite: jessie
Codename: jessie/mongodb-org
Architectures: amd64 i386
```

abhinavgupta, 2018-01-08:
@Fourdee Thanks, their testing branch has a binary for arm64, any recommendations on how to tackle this problem further?

bhaveshgohel, 2018-04-13:
Dear All,

I find https://andyfelong.com/2017/03/mongodb-3-0-14-binaries-for-raspberry-pi-3 MonogDB 3.0.14 binary to be working fine in bellow system:

Board: Odroid XU4
Architecture : armv7l GNU/Linux
DietPi: v6.6

Thanks Andy (@AndyOfLinux) for your work :+1: