MichaIng/DietPi #891: DietPi-Software | Support H3 GPU + Kodi

Fourdee, 2017-04-21:
> http://dietpi.com/phpbb/viewtopic.php?f=9&t=1642&p=6991#p6990

Invictaz, 2017-05-07:
The library of @Mosterta has some good SunXi things for H3 

Github.com/mosterta

Fourdee, 2017-05-12:
@igorpecovnik @zador-blood-stained 

Apologies for bringing you guys in.

When building ARMbian **server** images (H3), are there any specific changes related to GPU that would effect operation, once drivers are built/installed?
eg:
- Kernel options/modules reduction for server image build
- Command line option to disable mali memory share

zador-blood-stained, 2017-05-12:
On current H3 legacy server images memory reservations are disabled by default via kernel arguments (although not all related kernel arguments take effect on H3 kernel). Compile-time configuration does not change between server and desktop images.

Memory reservations can be enabled by adding `disp_mem_reserves=on` to `/boot/armbianEnv.txt`: https://github.com/armbian/build/blob/master/desktop.sh#L66

Fourdee, 2017-05-13:
Sorry guys, going to have to nudge this out of v150. Due to limited time at the moment to work on this in depth, initial unsuccessful attempts with available online sources, and, holding up v150 release.

I'll pick this back up when I can.

closed 2017-12-30 by Fourdee