MichaIng/DietPi #893: DietPi-Software | XU4 + Kodi. Black screen.

Fourdee, 2017-04-21:


mims92, 2017-04-21:
My reference code: ca4267a7-4228-41ef-8c23-c4b3ba165aa2-0

Fourdee, 2017-04-22:
Unable to replicate on test system. Will check bug report.

Fourdee, 2017-04-22:
```
[  4971.619] (WW) xf86CloseConsole: VT_GETMODE failed: Input/output error
[  4971.619] (EE) 
Fatal server error:
[  4971.619] (EE) xf86CloseConsole: VT_ACTIVATE failed: Input/output error
```

@mims92 

> Kodi starts but I have a black screen. I can 'hear' Kodi menu when pressing arrow keys.

Lets try EDID display modes, in ```/DietPi/dietpi.txt```, change existing entry ``` setenv videoconfig``` to line below
```
 setenv videoconfig "video=HDMI-A-1:1920x1080@60" 
```

And force HDMI plugged in:
```
setenv HPD "false"
```

mims92, 2017-04-22:
Still a black screen.

Here is the log of /var/log/Xorg.0.log
```
502.776]
X.Org X Server 1.16.4
Release Date: 2014-12-20
[   502.785] X Protocol Version 11, Revision 0
[   502.788] Build Operating System: Linux 3.16.0-0.bpo.4-armmp-lpae armv7l Debian
[   502.791] Current Operating System: Linux mims-droid 3.10.104+ #1 SMP PREEMPT Tue Feb 21 14:20:54 CET 2017 armv7l
[   502.791] Kernel command line: console=tty1 root=/dev/mmcblk0p2 rootwait ro fsck.repair=yes drm_kms_helper.edid_firmware=edid/1920x1080.bin smsc95xx.macaddr=00:1e:06:61:7a:72 governor=performance hdmi_tx_amp_lvl=31 hdmi_tx_lvl_ch0=3 hdmi_tx_lvl_ch1=3 hdmi_tx_lvl_ch2=3 hdmi_tx_emp_lvl=6 hdmi_clk_amp_lvl=31 hdmi_tx_res=0 HPD=true vout=hdmi
[   502.801] Build Date: 11 February 2015  01:18:06AM
[   502.804] xorg-server 2:1.16.4-1 (http://www.debian.org/support)
[   502.807] Current version of pixman: 0.32.6
[   502.814]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[   502.814] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   502.827] (==) Log file: "/var/log/Xorg.0.log", Time: Sun Apr 23 01:08:57 2017
[   502.830] (==) Using config file: "/etc/X11/xorg.conf"
[   502.833] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   502.834] (==) No Layout section.  Using the first Screen section.
[   502.834] (**) |-->Screen "Default Screen" (0)
[   502.834] (**) |   |-->Monitor "<default monitor>"
[   502.834] (**) |   |-->Device "Mali-Fbdev"
[   502.834] (==) No monitor specified for screen "Default Screen".
        Using a default monitor configuration.
[   502.834] (**) Option "NoTrapSignals" "true"
[   502.835] (**) Option "DontZap" "false"
[   502.835] (**) Option "BlankTime" "0"
[   502.835] (**) Option "StandbyTime" "0"
[   502.835] (**) Option "SuspendTime" "0"
[   502.835] (**) Option "OffTime" "0"
[   502.835] (==) Automatically adding devices
[   502.835] (==) Automatically enabling devices
[   502.835] (==) Automatically adding GPU devices
[   502.835] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   502.835]    Entry deleted from font path.
[   502.835] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[   502.835]    Entry deleted from font path.
[   502.835] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.[   502.835]    Entry deleted from font path.
[   502.836] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[   502.836]    Entry deleted from font path.
[   502.836] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.[   502.836]    Entry deleted from font path.
[   502.836] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   502.836]    Entry deleted from font path.
[   502.836] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[   502.836] (==) ModulePath set to "/usr/lib/xorg/modules"
[   502.836] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[   502.836] (II) Loader magic: 0xb6fb4f18
[   502.836] (II) Module ABI versions:
[   502.836]    X.Org ANSI C Emulation: 0.4
[   502.836]    X.Org Video Driver: 18.0
[   502.836]    X.Org XInput driver : 21.0
[   502.836]    X.Org Server Extension : 8.0
[   502.837] (II) xfree86: Adding drm device (/dev/dri/card0)
[   502.838] (II) no primary bus or device found
[   502.838]    falling back to /sys/devices/platform/exynos-drm/drm/card0
[   502.838] (II) LoadModule: "glx"
[   502.838] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[   502.845] (II) Module glx: vendor="X.Org Foundation"
[   502.845]    compiled for 1.16.4, module version = 1.0.0
[   502.845]    ABI class: X.Org Server Extension, version 8.0
[   502.845] (==) AIGLX enabled
[   502.845] (II) LoadModule: "armsoc"
[   502.846] (II) Loading /usr/lib/xorg/modules/drivers/armsoc_drv.so
[... 131 lines cut ...]
[   503.155] (**) evdev: gpio_keys.7: Device: "/dev/input/event0"
[   503.155] (--) evdev: gpio_keys.7: Vendor 0x1 Product 0x1
[   503.155] (--) evdev: gpio_keys.7: Found keys
[   503.155] (II) evdev: gpio_keys.7: Configuring as keyboard
[   503.155] (**) Option "config_info" "udev:/sys/devices/gpio_keys.7/input/input0/event0"
[   503.155] (II) XINPUT: Adding extended input device "gpio_keys.7" (type: KEYBOARD, id 6)
[   503.155] (**) Option "xkb_rules" "evdev"
[   503.155] (**) Option "xkb_model" "trustda"
[   503.155] (**) Option "xkb_layout" "be"
[   505.083] (II) ARMSOC(0): EDID vendor "LNX", prod id 0
[   505.083] (II) ARMSOC(0): Using hsync ranges from config file
[   505.083] (II) ARMSOC(0): Using vrefresh ranges from config file
[   505.083] (II) ARMSOC(0): Printing DDC gathered Modelines:
[   505.083] (II) ARMSOC(0): Modeline "1920x1080"x0.0  148.50  1920 2008 2052 2200  1080 1084 1089 1125 +hsync +vsync (67.5 kHz eP)
[   505.083] (II) ARMSOC(0): Modeline "1920x1080"x60.0  172.80  1920 2040 2248 2576  1080 1081 1084 1118 -hsync +vsync (67.1 kHz e)
[   505.083] (II) ARMSOC(0): EDID vendor "LNX", prod id 0
[   505.083] (II) ARMSOC(0): Using hsync ranges from config file
[   505.083] (II) ARMSOC(0): Using vrefresh ranges from config file
[   505.083] (II) ARMSOC(0): Printing DDC gathered Modelines:
[   505.083] (II) ARMSOC(0): Modeline "1920x1080"x0.0  148.50  1920 2008 2052 2200  1080 1084 1089 1125 +hsync +vsync (67.5 kHz eP)
[   505.083] (II) ARMSOC(0): Modeline "1920x1080"x60.0  172.80  1920 2040 2248 2576  1080 1081 1084 1118 -hsync +vsync (67.1 kHz e)
[   507.306] (II) evdev: gpio_keys.7: Close
[   507.306] (II) UnloadModule: "evdev"
[   507.615] (EE) Server terminated successfully (0). Closing log file
```

Fourdee, 2017-04-23:
@mims92 

Thanks.

Lets check installed EGL, please paste results:
```
find /usr/lib -name libEGL*
```

It should match:
```
root@DietPi:~# find /usr/lib -name libEGL*
/usr/lib/libEGL.so.1.4
/usr/lib/arm-linux-gnueabihf/libEGL.so.1.4
/usr/lib/arm-linux-gnueabihf/libEGL.so.1
/usr/lib/arm-linux-gnueabihf/libEGL.so
/usr/lib/arm-linux-gnueabihf/libEGL.so.1.0.0
/usr/lib/libEGL.so.1
/usr/lib/libEGL.so
```
If it doesnt match, try purging/reinstalling:
```
apt-get purge xf86-video-armsoc-odroid malit628-odroid
apt-get install xf86-video-armsoc-odroid malit628-odroid
```

Fourdee, 2017-04-23:
@mims92 

You also have some additional mesa packages:
```
libgles2-mesa:armhf
libglu1-mesa:armhf        
```

Try removing these:
```
apt-get purge libgles2-mesa libglu1-mesa
```

mims92, 2017-04-23:
@fourdee

I ran `find /usr/lib -name libEGL*` and the output is the same.

Uninstalling 
```
libgles2-mesa:armhf
libglu1-mesa:armhf
```
would remove Kodi.

Here is my Xorg.conf:
```
# X.Org X server configuration file for xfree86-video-mali

Section "Device"
        Identifier      "Mali-Fbdev"
        Driver          "armsoc"
        Option          "fbdev"                 "/dev/fb0"
        Option          "Debug"                 "false"
        Option          "DPMS"                  "false"
        Option          "Fimg2DExa"             "true"
        Option          "DRI2"                  "true"
        Option          "DRI2_PAGE_FLIP"        "false"
#       Option          "Fimg2DExaSolid"        "false"
#       Option          "Fimg2DExaCopy"         "false"
        Option          "Fimg2DExaComposite"    "true"
        Option          "SWcursorLCD"           "false"
EndSection

Section "Screen"
        Identifier   "Default Screen"
        Device       "Mali-Fbdev"
        DefaultDepth 24
EndSection

Section "DRI"
        Mode 0666
EndSection
```

Fourdee, 2017-04-23:
@mims92 

> Here is my Xorg.conf:

Yep, thats fine: 
https://github.com/Fourdee/DietPi/blob/master/dietpi/conf/xorg_xu4.conf

All your installed packages/settings appear to be in order. I believe this may be a simple case of incompatibly with your TV/XU4. We can assume Kodi is running (as you mentioned menu sounds are functional)

Few things you could try to debug:
- Different HDMI cable
- Lower resolution (720p)
- Different TV
- Fresh installation of DietPi + Kodi selected

mims92, 2017-04-24:
I've an eMMc with Android installed on it it was fine. I can see the console on my TV as well.

I will try by lowering the resolution and maybe change some settings.

Reinstalling would take some times because of all the installed software (I'm lazy I know).

I will post my findings if any.

mims92, 2017-04-26:
On my rapsberry pi (dietpi also), it is working.
Same cable, same tv

mims92, 2017-04-30:
Hi @Fourdee,

I installed a fresh installation of dietpi + Kodi only, I don't have any error but still have a black screen.

I noticed a small square window with a message (maybe not important):
```
libEGL warning: DRI2: failed to authenticate
```
before having the black screen.

Fourdee, 2017-05-08:
@mims92 

Sorry to hear this is still an outstanding issue.
However, this is the only reported issue for XU4 Kodi + black screen from our users, and, we cant replicate it.

> I've an eMMc with Android installed on it it was fine. I can see the console on my TV as well.

- If it works on Android, it may be a feature compiled into the Android Kodi binary, that is lacking in the debian one. 
- As none of the known HDMI settings/tweaks worked (https://github.com/Fourdee/DietPi/issues/893#issuecomment-296372835), it could be the Android kernel is somehow more compatible with your TV.

It could be anything, however, being honest, i'am not sure what the cause is.

> I installed a fresh installation of dietpi + Kodi only, I don't have any error but still have a black screen.

It may be worth testing various HDMI settings in ```boot.ini``` on this fresh install aswell:
https://github.com/Fourdee/DietPi/issues/893#issuecomment-296372835