MichaIng/DietPi #894: DietPi-Software | DB preference (MySQL/MariaDB)

Fourdee, 2017-04-21:
Similar to Webserver preference option:
> http://dietpi.com/phpbb/viewtopic.php?f=9&t=1814&p=6999#p6970

CallumH003, 2017-04-21:
Cheers @Fourdee ! Seems like something that could be beneficial to DietPi users :)

MichaIng, 2017-06-02:
+1
Chose a webserver stack now (LAAP) to assure that mariadb will be used as db, but now some error occurred on rpi stretch beta: dietpi-services shows mariadb AND mysql and tries to handle (start/stop/restart) both of them, where mariadb fails. Nevertheless the mysql service actually IS mariadb, which can be seen through version info, nextcloud webui etc.
This error might be caused by choosing stack + nextcloud, which again depends on, and possibly installs db again? Will add this to existing rpi stretch issue.

MichaIng, 2017-12-08:
@Fourdee 
Close, as the effort does not make any sense anymore, where MySQL installation is not possible through Stretch repo.
Migration is possible and simple: https://mariadb.com/de/node/546
We could force that through patchfile or encourage people to shift.

Fourdee, 2017-12-08:
@MichaIng 

> And there are other databases, like postresql, we could add.

Yep, i'll leave it open for now, however, this feature will only work with full SQL compatible software, that could replace MariaDB for our software installations. 
SQlite is its own installation.

MichaIng, 2018-01-30:
@Fourdee 
I close, as this does not make sense at the moment without MySQL. I don't know any software where we offer a choice by installation/configuration between SQLite and MariaDB. Most need either the one or the other. Even that e.g. ownCloud and Nextcloud work with both, MariaDB is clearly the only viable production solution.

We can rethink after re rethought and integrated e.g. postgreSQL 😄.

closed 2018-01-30 by MichaIng