MichaIng/DietPi #896: General | Change default user to dietpi, including all software installs

Fourdee, 2017-04-22:
currently ```root```

> https://github.com/Fourdee/DietPi/issues/796#issuecomment-296365067

sootstone, 2017-06-27:
Would it be possible to make username changeable in dietpi.txt on first install?

Fourdee, 2017-06-27:
@sootstone 

> Would it be possible to make username changeable in dietpi.txt on first install?

Good idea 👍 , i'll see what we can do.

Fourdee, 2017-12-30:
We'll attempt to achieve this for v6.0, however, we have a limit of 1 month for completion with many other features yet to implement, so this may be a continuing process in v6.0's future.

Fourdee, 2018-01-09:
Not enough time to make this into v6.0, and, is low priority.

Moving to v6.1

MichaIng, 2018-01-09:
Agree, also I would make it optional, or default, but optional to just stay with root. Generally I agree to use non-root user for default actions, but on otherwise secure/updated servers with just ports 80+443 forwarded, limited hardened web services there and for local access, strong password + SSH key authentication, I would clearly prefer the comport advantages of by default using root for server maintenance.

But all our scripts should just work the same, if started with sudo, right?

Fourdee, 2018-02-03:
@MichaIng 

> But all our scripts should just work the same, if started with sudo, right?

Yep, ```sudo``` elevates to ```root``` user. Effectively the same as logging in as root.

Fourdee, 2018-06-29:
Currently, following has been implemented:
- Use of, and support for ```dietpi``` user on SSH/local terminal login and usage

Pending:
- Desktops/VNC as ```dietpi``` user?
- Implement unique users for each software installation title that requires it (different ticket: https://github.com/Fourdee/DietPi/issues/1877)

Fourdee, 2018-07-09:
Marking as closed, in favor of: https://github.com/Fourdee/DietPi/issues/1877

closed 2018-07-09 by Fourdee