MichaIng/DietPi #899: XU4 | Prep for Meveric's repo changing to 4.9 kernel

Fourdee, 2017-04-26:


joaofl, 2017-04-27:
Hi Fourdee. Can I help on something with this task?

I may have some free time soon, maybe to test this and improve that config dir thing we discussed about syncthing.

Fourdee, 2017-04-27:
@joaofl 

> Hi Fourdee. Can I help on something with this task?

By all means, thanks for offering 👍 
Once i figure out how will we patch uuid in ```boot.ini``` to replace ```/dev/mmcxxxx````, should be good to test. 
I'll also create a new image with the kernel installed, for testing.

Fourdee, 2017-04-29:
Notes:

- Get UUID of p1: ```blkid | grep -m1 '^/dev/mmcblk0p1' | awk '{print $4}' | sed 's/.*=//' | sed 's/"//g'```

Fourdee, 2017-04-29:
Weird, if serial UART is connected to SBC pins and USB UART pins, but not connected to a USB socket, system fails to boot.

Fourdee, 2017-04-29:
@joaofl 

Any chance you could test this (and ethernet status) on your XU4?

To install 4.9 kernel, as per https://github.com/Fourdee/DietPi/issues/899#issuecomment-298170796:
- Run code in ####Patch Notes
- Then code in #### Install kernel test:

joaofl, 2017-04-30:
@Fourdee 

My first test was with my current setup, in which I already had the UUID set on both files (fstab and boot.ini) since I was booting from an USB pendrive.
However it failed to boot. It halts at: 
`Apr 30 02:08:37 DietPi systemd[427]: Failed at step EXEC spawning /bin/plymouth: No such file or directory`
(journal file attached).
[journal.txt](https://github.com/Fourdee/DietPi/files/966512/journal.txt)

My second test was on a new install:
It halts at that same plymouth issue. But there are other errors concerning the mmcmblk1 (timed out). See the journal attached.
[journal-freshinstall.txt](https://github.com/Fourdee/DietPi/files/966516/journal-freshinstall.txt)
.

Fourdee, 2017-04-30:
@joaofl 

Thanks for testing.

> Is there an easy way to rollback to the previous kernel and fix this issue?

Yep, if you copy the contents of ```/boot``` on the image file, to SD card/EMMC, excluding:
- dietpi.txt
- boot.ini
- /dietpi (folder)

And modules from image:
- ```/lib/modules```

Should boot back up

Fourdee, 2017-04-30:
Possible solutions regarding ```Failed at step EXEC spawning /bin/plymouth: No such file or directory``` 
https://unix.stackexchange.com/a/178640

If you test again, before rebooting after kernel update, we should probably check patched ```/etc/fstab``` and ```blkid``` match:
```
blkid
cat /etc/fstab
```

Fourdee, 2017-04-30:
🈯️ Network bandwidth much improved over 3.x (was around 400Mbit):
```
Client connecting to 192.168.0.112, TCP port 5001
TCP window size: 87.5 KByte (default)
------------------------------------------------------------
[  5] local 192.168.0.111 port 51702 connected with 192.168.0.112 port 5001
[ ID] Interval       Transfer     Bandwidth
[  5]  0.0-30.0 sec  3.28 GBytes   938 Mbits/sec
[  4] local 192.168.0.111 port 5001 connected with 192.168.0.112 port 38925
[  4]  0.0-30.0 sec  2.68 GBytes   766 Mbits/sec
```

joaofl, 2017-04-30:
@Fourdee 
Indeed, by commenting two custom mounts I had on my fstab, it booted. Buuuut, without ethernet.

On dmesg, it shows that it failed to boot some modules  for example the s5p which was not found. also fuse, IPV6... I guess the first is from the GPU that in fact I did not install since I dont use it currently.

How did you get your ethernet working? I remember someone mentioned this issue in some other forum, I'll try to figure it out.

Fourdee, 2017-04-30:
@joaofl 

For XU3/4, dtb's are now unique. So if your running a XU4:
```
sed -i "s#exynos5422-odroidxu3.dtb#exynos5422-odroidxu4.dtb#g" /DietPi/boot.ini
reboot
```

Fourdee, 2017-04-30:
@joaofl 

> I also noticed that Dietpi fails to fetch CPU governor settings...

Interactive is not available in 4.9, ondemand and performance govs only:
[image]

DietPi should cycle through govs and re-apply, if current one does not exist.

joaofl, 2017-04-30:
@Fourdee 
Thanks! It works now. Buuut, :) Cloudshell stopped working... Is the SPI module loaded? I could help on testing that as well.

joaofl, 2017-04-30:
@Fourdee 

Is the CPU limited to 1.3 GHz? 2.0 doesn't show in the list of frequencies available.

Some minor issues:

- When I chose to change the "On demand sampling rate" I get the error:
```
cat: /sys/devices/system/cpu/cpufreq/ondemand/sampling_rate_min: No such file or directory
/DietPi/dietpi/dietpi-config: line 1424: / 1000 : syntax error: operand expected (error token is "/ 1000 ")
```

Fourdee, 2017-05-01:
@joaofl 

> Is the CPU limited to 1.3 GHz?

Yep, for cores 0-3, other cores are still 2GHz. They must of tweaked max frequencies a little to try to reduce known heat issues with XU4.

> but when transferring a file through sftp, I get at most 220 Mbps (28 MB/s). I believe the bottleneck in this case is the CPU. What do you think?

Good find, new location is:
```
/sys/devices/system/cpu/cpufreq/ondemand/min_sampling_rate
```

Fourdee, 2017-05-03:
ToDo:
- Test if ``` option "NoFlip" "true" ``` is compatible with 3.x
- Add whiptail option for user to select XU3/4 board during patch: https://github.com/Fourdee/DietPi/issues/899#issuecomment-298234706

Fourdee, 2017-05-03:
- 🈯️ XU4 4.9 image, for TESTING purposes only:
http://dietpi.com/downloads/testing/DietPi_OdroidXU4-armv7-(Jessie).7z

--- 

note to self, 99MB, check ```/lib/modules/3.x``` is removed.

Fourdee, 2017-05-05:
@rncwnd79 

>  because /proc/cpuinfo doesn't contain the odroid-string anymore:

Thanks for the report. 👍 

You can avoid the string check and force the HW_MODEL index, for XU4:
```
echo 11 > /etc/.dietpi_hw_model_identifier
```

I'll add that by default to our next XU4 image and remove string check, to avoid this.

Fourdee, 2017-05-05:
- 🈴 Seems to have no effect. Kodi is extremely jittery in 4.9 with updated GPU driver:
```
    Option      "NoFlip" "true"
```

Fourdee, 2017-05-05:
Ok:
- 🈯️ Prep for 4.9 kernel is now completed during v150 patch
- 🈯️ 4.9 kernel will not be installed during v150, we'll wait for Meveric to release 4.9 through apt (we can then assume stable for end users)
- 🈺 We'll offer a 4.9 kernel test image for our users (needs updating)

Fourdee, 2017-05-05:
Will mark this ticket as completed. Open up a new one for further 4.9 testing:
https://github.com/Fourdee/DietPi/issues/926

closed 2017-05-05 by Fourdee