MichaIng/DietPi #906: "su" isn't working

greencopper, 2017-04-28:
I have just downloaded and installed DietPi on an ODROID XU4 and everything works perfectly except for "su".

I log in to the box using OpenSSH. Afterwards I do "su" to become root. I get the password prompt, but even though the correct password is entered, it fails with an authentication error.

Logging in as root using SSH works.

Kind regards

Fourdee, 2017-04-28:
@greencopper 

Thanks for the report 👍 

I've got a few XU4 jobs to complete over the next few days, i'll check this at the same time.

Fourdee, 2017-05-01:
@greencopper 

>  Afterwards I do "su" to become root. I get the password prompt, but even though the correct password is entered, it fails with an authentication error.

Did you add your user to group ```sudo```?

If your running DietPi v149, try our test DietPi user, see if you get same issue
- username = dietpi
- password = dietpi

greencopper, 2017-05-02:
Adding the user to sudo should make no difference as "su" and "sudo" aren't the same.

In my case I remove sudo, if it's installed, I only want "su" to work.

"su" normally works out-of-the-box on Debian.

Fourdee, 2017-05-02:
@greencopper 

```
sudo su
```

Fourdee, 2017-05-06:
Marking as resolved using:
https://github.com/Fourdee/DietPi/issues/906#issuecomment-298673747

closed 2017-05-06 by Fourdee