MichaIng/DietPi #907: mount: unknown filesystem type 'xfs'

greencopper, 2017-04-29:
I have installed xfsprogs and wanted to mount a XFS drive.

```
# mount -t xfs /dev/sdb1 /mnt/
mount: unknown filesystem type 'xfs'
```

I suspect this is a custom kernel and the XFS module hasn't been compiled in?

```
# modprobe xfs
modprobe: FATAL: Module xfs not found.
```

Kind regards.

Fourdee, 2017-04-29:
@greencopper 

Thanks for the report.

The kernel we use on XU4 is standard Odroid kernel , provided and maintained by Meveric (with possibly some minor tweaks).

I'll send Meveric a message with your request for XFS support.

Fourdee, 2017-04-30:
@greencopper 

Quote from Meveric:
> Just looked through the forum. It seems enabling XFS breaks docker support in the Kernel.
But Kernel 4.9 might have activated XFS.

Seems like a no go for XFS on XU4 currently.

But it does work in 4.9 kernel:
https://github.com/Fourdee/DietPi/issues/899
```
root@DietPi:~# modprobe xfs
root@DietPi:~# lsmod
Module                  Size  Used by
xfs                   680632  0
libcrc32c                935  1 xfs
ads7846                11687  0
spidev                  6446  0
spi_s3c64xx            11643  0
exynos_gpiomem          3162  0
fuse                   75590  1
ipv6                  341326  28
uas                    11815  0
root@DietPi:~# uname -a
Linux DietPi 4.9.24+ #1 SMP PREEMPT Fri Apr 28 17:53:43 UTC 2017 armv7l GNU/Linux
```

greencopper, 2017-04-30:
Is there a working image for that?

greencopper, 2017-04-30:
Sorry, found it.

http://linuxfactory.or.kr/dokuwiki/doku.php?id=yet_another_debian_ubuntu_release_for_odroid-xu4&&ckattempt=2

Fourdee, 2018-04-05:
Marking as closed due to:
- Kernel (outside of our support range)

closed 2018-04-05 by Fourdee