MichaIng/DietPi #908: mount: wrong fs type, bad option, bad superblock on /dev/sda1

greencopper, 2017-04-29:
I wanted to have a seagate disk spin down, but due to the journal settings of ext4 the drive continuously works.

On way to overcome this is by using XFS as a filesystem, but since that isn't working, I did the following:

```
# mkfs.ext4 -E lazy_itable_init=0,lazy_journal_init=0 /dev/sdXY
```

This works and makes the disk spin down. However, now it cannot be mounted on the XU4.

```
# mount /pub1/
mount: wrong fs type, bad option, bad superblock on /dev/sda1,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
```

If I mount the disc on my regular Linux distro then there isn't a problem.

Kind regards.

Fourdee, 2017-04-29:
dupe of:
https://github.com/Fourdee/DietPi/issues/907#issue-225224383

greencopper, 2017-04-29:
This is not a dupe :) #907 is about XFS. In the above I have tried using EXT4, but with "lazy_itable_init=0,lazy_journal_init=0" setting.

Fourdee, 2017-04-29:
@greencopper 

> This is not a dupe 

Quite right, apologies :)

Lets check partition layout (please paste results):
```
blkid
```

And after mounting the drive (with failure), lets check kernel error messages:
```
dmesg
```

^^ the above will be a large printout, but we only need stuff after ```[13.xxx]```.

greencopper, 2017-04-29:
No, problem ;)

```
# blkid
/dev/mmcblk0: PTUUID="000f1766" PTTYPE="dos"
/dev/mmcblk0p1: SEC_TYPE="msdos" LABEL="boot" UUID="5886-8548" TYPE="vfat" PARTUUID="000f1766-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="a4f12232-1670-43f4-8eca-49bfbe4d01ee" TYPE="ext4" PARTUUID="000f1766-02"
/dev/sda1: UUID="cef0bf49-2656-49c5-9990-ecf8c1aef801" TYPE="ext4" PARTUUID="164c3a78-57fc-45e3-a33e-60e158b27590"
´´´

And the dmesg part I think you want, after a failed mount:

```
[  121.443517] [c4] JBD2: Unrecognised features on journal
[  121.443531] [c4] EXT4-fs (sda1): error loading journal
```

Kind regards.

greencopper, 2017-05-01:
I am sorry, but there is no issue. The problem was in my fstab, I didn't notice and forgot to check the log.

The disk mounts perfectly now.

greencopper, 2017-05-01:
Turns out there actually is a problem, periodically.

I had formatted the disk on another computer with a brand new kernel and stuff. The ext4 module from the DietPi with kernel 3.10 is too old and does not recognize the ext4 from the new.

If I format the drive on the ODROID XU4 with the old kernel and leave it attached, then I can use it without the problems above.

Is there any way to get a never kernel running with DietPi that doesn't need a lot of hacking?

Fourdee, 2017-05-02:
@greencopper 
If you are running XU4, you can test the 4.9 kernel in this post:
https://github.com/Fourdee/DietPi/issues/899#issuecomment-298170796

Bear in mind, we are still making fixes in DietPi to support this new kernel (see link above), but in general, seems stable.

Fourdee, 2017-06-17:
No updates/responses, assuming resolved. Please reopen if required.

closed 2017-06-17 by Fourdee