MichaIng/DietPi #909: CPU Temperature error

hectorcamp, 2017-04-30:
OrangePi with the new update v149, shows the CPU temperature as 0 °
[image]

Fourdee, 2017-04-30:
@hectorcamp 

Thanks for the report 👍 

We'll get that fixed :)

Fourdee, 2017-04-30:
OPi PC report: http://dietpi.com/phpbb/viewtopic.php?f=11&t=1847#p7076

Fourdee, 2017-04-30:
Confirmed:
- OPi one
- NanoPi M1

Seems all H3's use 2 digit temp readouts.

Fourdee, 2017-04-30:
@hectorcamp 

Think we fixed it, any chance you can update script and test for us?
```
wget https://github.com/Fourdee/DietPi/raw/testing/dietpi/dietpi-cpuinfo -O /DietPi/dietpi/dietpi-cpuinfo
cpu
```

hectorcamp, 2017-04-30:
It works, thank you!.
[image]

jcjdominguez, 2017-04-30:
>Think we fixed it, any chance you can update script and test for us?

wget https://github.com/Fourdee/DietPi/raw/testing/dietpi/dietpi-cpuinfo -O /DietPi/dietpi/dietpi-cpuinfo
cpu

Hi. I also have this issue on a opi zero, but that fix doesn't work for me.

motaboy, 2017-05-02:
Same Problem on Orange Pi Zero with H2+ CPU.

Please fix if possible as cat temp shows the value properly.

motaboy, 2017-05-02:
DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     armv7l
 Temp         |     0'c | Who put me in the freezer!
 Governor     |     ondemand
 Throttle up  |     85% CPU usage

                 Current Freq    Min Freq   Max Freq
 CPU0         |      240 MHz      240 MHz    1200 MHz
 CPU1         |      240 MHz      240 MHz    1200 MHz
 CPU2         |      240 MHz      240 MHz    1200 MHz
 CPU3         |      240 MHz      240 MHz    1200 MHz

Fourdee, 2017-05-02:
@motaboy 

Thanks for the report 👍 

Does this fix work for you? https://github.com/Fourdee/DietPi/issues/909#issuecomment-298228895

motaboy, 2017-05-02:
No, tried it twice but didn't fix it.

Fourdee, 2017-05-02:
@motaboy 

Thanks.

I've updated script, please can you test it for us again?
https://github.com/Fourdee/DietPi/issues/909#issuecomment-298228895

motaboy, 2017-05-02:
Nice! 👌
Working fine now

Fourdee, 2017-05-03:
Thanks to everyone who reported and tested this for us.

I think we can say this is now resolved. Marking as completed. Please reopen if required,

closed 2017-05-03 by Fourdee

syl3nt, 2017-05-04:
I confirm the 
"wget https://github.com/Fourdee/DietPi/raw/testing/dietpi/dietpi-cpuinfo -O /DietPi/dietpi/dietpi-cpuinfo"
worked for me.
device:OPi PC

jonasmorais, 2017-05-06:
I confirm the
"wget https://github.com/Fourdee/DietPi/raw/testing/dietpi/dietpi-cpuinfo -O /DietPi/dietpi/dietpi-cpuinfo"
worked for me.
device:OPi One
thanks
Jonas

yo8aiv, 2017-05-08:
i confirm also this fix worked also for me.
 V149       | OrangePi Zero (armv7l)

Ahaus314, 2017-05-08:
Fix on my NanoPi M3!