MichaIng/DietPi #911: DietPi Image | OrangePi PC 2

pidiet, 2017-05-01:
Now that OrangePi Prime (same H5 SOC) is supported this one should be **easy**

_Tag:_ DietPi Image

osmandoandbackup, 2017-05-02:
Orange Pi PC 2 is the only single-board computer **with Gigabit-Ethernet** and 64bit CPU **under 20$**.

Best performance/price-ratio you can get today!

Fourdee, 2017-05-02:
I dont own a OPi PC 2 yet, but i should be able to generate the image and prep on a prime board. I'll take a look.

Fourdee, 2017-05-02:
Dev branch image, put packages on hold:
```
apt-mark hold linux-dtb-dev-sun50iw2 linux-image-dev-sun50iw2 linux-jessie-root-dev-orangepipc2 linux-u-boot-orangepipc2-dev
```

Fourdee, 2017-05-02:
- 🈯️ Image: http://dietpi.com/downloads/images/DietPi_OrangePiPC2-arm64-(Jessie).7z

Fourdee, 2017-05-03:
I believe this image will have the same lack of audio as per the Prime board:
https://github.com/Fourdee/DietPi/issues/874#issuecomment-297675707

If anyone can confirm this, i'd be grateful:
```
apt-get install alsa-utils
aplay -l
```

Fourdee, 2017-05-13:
I'll mark this as completed.

Please reopen if required.

closed 2017-05-13 by Fourdee

pidiet, 2017-08-09:
I found a bug (maybe it's know?)

**It's not possible to shut down the system**, not with 'poweroff' and not with 'init 0'. Only dirty work'a'round is to cut the powersupply.

Next thing I found is that right now kernel 4.10.0 is used.

Lot's of things got introduced in kernel 4.12 (for e.g Audio Codec, Clocks, DMA, I2C, Pinctrl, PWM, SD/MMC, SPDIF, SPI, SRAM, USB, OTG, Watchdog) and the best is coming with 4.13 (right now RC) which should be proper Ethernet support (EMAC & GMAC)! Some as these also apply to Allwinner H3 not only H5 devices!

More @ linux-sunxi: https://linux-sunxi.org/Linux_mainlining_effort#Merged_into_4.13

Fourdee, 2017-08-10:
@pidiet 

We use ARMbian to build this image with their provided kernel. Inability to power off is a known issue:
https://www.armbian.com/orange-pi-pc2/

Kernel updates are provided by ARMbian's repo, so we must wait for those to be switched to 4.12+.

pidiet, 2017-08-10:
Ah, OK! Good to know! Kernel 4.13 should be released in the beginning of September.. so let's see what comes ;)

The download archive from Armbian (https://dl.armbian.com/orangepipc2/archive/) is providing Kernel 4.11.1 for OPiPC2 and the nightly's also. Maybe there is a regression in updating?

kornfool, 2017-08-23:
having problem on mounting drive with ,iocharset=utf8 with this setting it get error, without this all text was not shown properly. orange pi pc 2