MichaIng/DietPi #915: DietPi-Software | PiCast

Fourdee, 2017-05-03:
> https://github.com/lanceseidman/PiCAST

Fourdee, 2017-05-03:
In short, PiCast is not Chromecast, or in any way compatible. Its current abilities are playing youtube videos from a URL link, that must be inputted via a browser.
[image]

Its essentially a highly glorified youtube video player, with no relation to Chromecast.

Chromecast is currently closed source/proprietary, hence why we dont see any real support for it, outside android and Google's own hardware.

The developer of PiCast has clearly incorrectly abused the word "Chromecast" to pull in interest and support, where its not due.

closed 2017-05-03 by Fourdee

Invictaz, 2017-05-03:
There are more https://github.com/spygi/personal/blob/master/raspberry-youtube/server.py