MichaIng/DietPi #917: Apt-get has failed

Sod-Almighty, 2017-05-03:
Apt-get has failed:                                                │
                         │  - Apt string = mate-desktop-environment-extras upower policykit-1 │
                         │ iceweasel p7zip-full --no-install-recommends                       │
                         │  - Error code = 100                                                │
                         │  - Software Title = MATE: desktop enviroment                       │
                         │  - Apt log = /var/log/dietpi-software_apt.log                      │
                         │                                                                    │
                         │ Please report this issue to DietPi so it can be investigated.      │
                         │                                                                    │
                         │ DietPi-Software will now exit.

Sod-Almighty, 2017-05-03:
> You don't have enough free space in /var/cache/apt/archives/

Should probably have a test for this.

Fourdee, 2017-05-04:
@Sod-Almighty 

> Should probably have a test for this.

We do in the next update v150, a general check for free space on RootFS:
You must have 500MB free space on RootFS, before ```dietpi-software``` and ```dietpi-update``` can be used.
https://github.com/Fourdee/DietPi/issues/905#issuecomment-298343932

Fourdee, 2017-05-04:
I'll mark this as closed (due to fix in https://github.com/Fourdee/DietPi/issues/905#issuecomment-298343932). Please reopen if required.

closed 2017-05-04 by Fourdee