MichaIng/DietPi #918: DietPi-LogClear | PiHole 3.0 now uses rolling logs

Fourdee, 2017-05-04:
- Test
- Without ```logrotate```?

WaLLy3K, 2017-05-04:
As long as `/var/log/pihole.log.1` can be exempt from the 1hr/24hr clearing of logs, that would be all that's needed from Diet Pi's end. If that's not an option, I'll be happy to change my logging method! :)

Fourdee, 2017-05-04:
@WaLLy3K 

Should be fine, i'll need to run some tests and see whats actually been changed for PiHole logging, then we can sort out a fix our end :)

Fourdee, 2017-05-04:
New password prompts during install. Seems they broke automated password entries:
```
::: Options:
:::  -p, password               Set web interface password, an empty input will remove any previously set password

root@DietPi:~# pihole -a -p dietpi
Enter New Password (Blank for no password):
```

Fourdee, 2017-05-04:
#### Notes:
- ```/var/log/pihole-FTL.log``` doesn't contain any data read by interface and can be cleared hourly.
```
[2017-05-04 13:09:01.604] Reading from /var/log/pihole.log (rwxrwxr-x)
[2017-05-04 13:09:01.604] Notice: Increasing queries struct size from 0 to 10000 (320.08 KB)
[2017-05-04 13:09:01.604] Notice: Increasing overTime struct size from 0 to 100 (324.08 KB)
[2017-05-04 13:09:01.604] Notice: Increasing forwarded struct size from 0 to 4 (324.19 KB)
[2017-05-04 13:09:02.473] New forward server: 8.8.4.4 google-public-dns-b.google.com (0/4)
[2017-05-04 13:09:02.473] Notice: Increasing domains struct size from 0 to 1000 (356.23 KB)
[2017-05-04 13:09:02.473] Notice: Increasing clients struct size from 0 to 10 (356.49 KB)
[2017-05-04 13:09:02.473] New client: 127.0.0.1 localhost (0/10)
[2017-05-04 13:09:02.509] New forward server: 8.8.8.8 google-public-dns-a.google.com (1/4)
[2017-05-04 13:09:02.513] New client: 192.168.0.23 (1/10)
[2017-05-04 13:09:02.516] Finished initial log file parsing
[2017-05-04 13:09:02.516]  -> Total DNS queries: 38
[2017-05-04 13:09:02.516]  -> Cached DNS queries: 0
[2017-05-04 13:09:02.516]  -> Blocked DNS queries: 8
[2017-05-04 13:09:02.516]  -> Unknown DNS queries: 0
[2017-05-04 13:09:02.516]  -> Unique domains: 34
[2017-05-04 13:09:02.516]  -> Unique clients: 2
[2017-05-04 13:09:02.516] Successfully accessed setupVars.conf
[2017-05-04 13:09:02.517] Listening on port 4711
```
- After log clears, restart ```pihole-FTL``` also.

Fourdee, 2017-05-04:
Above commit should resolve it. But i'll leave mine running over night and run some tests tomorrow to verify.

pidiet, 2017-05-04:
> 3.0.1
> Fixes 
> 
> Please check your admin interfaces and add a password if needed.  We have fixed the issue below, but please check your existing installs.  You can set a password with:
> 
> pihole -a -p somepassword
> **Fresh installs of v3.0 did not set a password on the Web interface despite the script providing the user with a password.  This has now been fixed.**

https://pi-hole.net/2017/05/04/pi-hole-3-0-1-fixes-and-tweaks/

Fourdee, 2017-05-05:
DietPi-RAMlog now supports multiple ```pihole.log*``` files.

> As long as /var/log/pihole.log.1 can be exempt from the 1hr/24hr clearing of logs,

@WaLLy3K 
With these changes, we apply the same 24hour clear for ```pihole.log``` and all other rolling logs (eg: ```pihole.log.1```). So any entries not matching current day, will be cleared hourly. We also have a 5MB limit on these files, to prevent RAMlog from using up its 20MB limit.

I'am yet to see a ```pihole.log.1``` file be generated in our testing. By any chance do you have logrotate installed? As this would cause rolling logs and isnt needed/supported by DietPi-RAMlog as we treat all log files (including rolling) the same.
```
dpkg -l | grep logrotate
```

WaLLy3K, 2017-05-06:
`pihole.log.1` should appear the midnight after an updated copy of Pi-hole has been installed.

Also:

```
w3k@MCT:~$ dpkg -l | grep logrotate
w3k@MCT:~$ which logrotate
w3k@MCT:~$ 
```

Fourdee, 2017-05-06:
@WaLLy3K 

> pihole.log.1 should appear the midnight after an updated copy of Pi-hole has been installed.

Legend, thank you 👍 

> As long as /var/log/pihole.log.1 can be exempt from the 1hr/24hr clearing of logs, that would be all that's needed from Diet Pi's end. If that's not an option, I'll be happy to change my logging method! :)

Ok, I got this completely wrong (2hours sleep + too much caffeine = bad idea), we didn't need to make those changes to support rolling logs with RAMlog, as we offer a alternative logging mode for our users.

#### If you want untouched logs, "Full logging mode" will do that:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p68

I'll revert some of these changes (keeping the FTL log changes as they are only for debugging).

Fourdee, 2017-05-06:
Completed.

k-plan, 2017-05-06:
@WaLLy3K 

> pihole.log.1 should appear the midnight after an updated copy of Pi-hole has been installed.

> If that's not an option, I'll be happy to change my logging method! :)

What does `pihole.log.1` contains and for which PiHole funktion we needed this?

@Fourdee 

> we didn't need to make those changes to support rolling logs with RAMlog, as we offer a alternative logging mode for our users.

Hmm, do we need to change `Logging System Choices` to `Full`, if we want to use a PiHole full feature system?
Until PiHole v.2x there was no different in PiHole web interface, if we use `DietPi-Ramlog #1` or `DietPi-Ramlog #2` or `Full`.

[image]

Sorry, it's not clear for me, what going on with `pihole.log.1`

WaLLy3K, 2017-05-07:
Glad I could help @Fourdee :)

> What does pihole.log.1 contains and for which PiHole funktion we needed this?

`pihole.log.1` is the renamed content of `pihole.log` from the previous day (which in turn, is the output log of `dnsmasq`, required for the Admin Console to display stats). Days prior to yesterday are stored as `pihole.log.#.gz`, and are no longer needed.

> Hmm, do we need to change Logging System Choices to Full, if we want to use a PiHole full feature system?

It does make me a bit sad that I need to have full blown logging to make use of the rolling 24 hour window. What about "DietPi-Ramlog 2" providing a symlink `/var/log/pihole.log.1 => /root/logfile_storage/pihole.log.1`? At least then one could have a cron scheduled to delete everything else in `logfile_storage`. I'd create a symlink myself, but I assume it'd get deleted after an hour.

Fourdee, 2017-05-07:
@WaLLy3K @k-plan 

Things we need to confirm:
- Does PiHole rotate these log files, without ```logrotate```? I'am yet to experience it in testing.
- Does PiHole (web stats) read data from the rotated log files?

Regardless of the above, DietPi RAMlog will only support 24 hour (current day) PiHole logs, with 5MB limit. We need to limit RAM usage.

Fourdee, 2017-05-07:
> What does pihole.log.1 contains

As per @k-plan , I still dont believe PiHole is rotating these logs with DietPi-RAMlog 1/2 (as ```logrotate``` is not installed), @WaLLy3K was there any contents in this file?

> pihole.log.1 should appear the midnight after an updated copy of Pi-hole has been installed.

I'll try to confirm this with testing.

Fourdee, 2017-05-07:
@WaLLy3K 

> It does make me a bit sad that I need to have full blown logging to make use of the rolling 24 hour window. What about "DietPi-Ramlog 2"

We still need to confirm the cause of the rolling log, and how/if it is generated without ```logrotate``` under DietPi-RAMlog 1/2.

If you want to keep existing daily logs for PiHole, I can upgrade DietPi-RAMlog mode 2, to backup before deletion in ```/root/logfile_storage/pihole.log```?

k-plan, 2017-05-07:
@Fourdee 
Thanks for reopened this ticket. 👍 

@WaLLy3K 

> `pihole.log.1` is the renamed content of `pihole.log` from the previous day (which in turn, is the output log of dnsmasq, required for the Admin Console to display stats).

Ahh, thanks for explanation. 👍  
Have play around a little bit like Fourdee do, build a `pihole.log.1` file and copy it to `/var/log/pihole.log.1`, but didn't see any changes in AdminLTE.

Is it possible, to show us a picture of AdminLTE with working `rolling log` /  `pihole.log.1` please?

> If you want to keep existing daily logs for PiHole, I can upgrade DietPi-RAMlog mode 2, to append before deletion in `/root/logfile_storage/pihole.log`?

Sound good. I want to have a "full feature PiHole" AdminLTE with `RAM-Log #2` to expand SD-Card lifespan. On a typical SBC PiHole installation, 5MB limit should not be a problem and I don't needed any history as `pihole.log.#.gz`.

Fourdee, 2017-05-08:
Ok, think its time we closed this ticket.

- In all testing, PiHole never generated a rotating log. Its done by ```logrotate``` package (if installed).

> pihole.log.1 should appear the midnight after an updated copy of Pi-hole has been installed.

- Unable to replicate this, however, even if this was added to PiHole, the PiHole update is outside of DietPi's control.

#### We will continue to support:
- 24 hour PiHole logs in DietPi-RAMlog 1/2 for ```/var/log/pihole.log```
- 5MB limit on PiHole logs in DietPi-RAMlog 1/2 for ```/var/log/pihole.log```
- Backups of ```/var/log/pihole.log``` in DietPi-RAMlog mode 2: https://github.com/Fourdee/DietPi/issues/933

@WaLLy3K @k-plan 
Thanks guys 👍 

Please reopen if required.

closed 2017-05-08 by Fourdee

k-plan, 2017-05-08:
With DietPi-RAMlog mode FULL:

[image]

```
root@RPi-Zero-W:~# ls -lah /var/log/ | grep pihole
-rw-r--r--  1 pihole   pihole   6,5K Mai  8 11:55 pihole-FTL.log
-rwxrwxr-x  1 dnsmasq  www-data 1,9M Mai  8 18:47 pihole.log
-rwxrwxr-x  1 dnsmasq  www-data 1,4M Mai  8 00:00 pihole.log.1

root@RPi-Zero-W:~# ls -lah /var/log/ | grep syslog
-rwxrwxr-x  1 root     adm       65K Mai  8 18:39 syslog

```

```
root@RPi-Zero-W:~# cat /var/log/pihole-FTL.log | grep 00:00
[2017-05-08 00:00:01.517] NOTICE: pihole.log has been flushed
[2017-05-08 00:00:01.517]   Resetting internal data structure
[2017-05-08 00:00:01.517]   Queries in memory before flushing: 4960
[2017-05-08 00:00:01.857] Gravity list entries: 117507
[2017-05-08 00:00:01.857] No blacklist present
[2017-05-08 00:00:01.857] No wildcard blocking list present
[2017-05-08 00:00:01.857] Reading from /var/log/pihole.log.1 (rwxrwxr-x)
[2017-05-08 00:00:01.858] Notice: Increasing queries struct size from 0 to 10000 (320.16 KB)
[2017-05-08 00:00:01.858] Notice: Increasing overTime struct size from 0 to 100 (323.36 KB)
[2017-05-08 00:00:01.859] Notice: Increasing forwarded struct size from 0 to 4 (323.44 KB)
[2017-05-08 00:00:01.869] New forward server: 192.168.1.100 (0/4)
[2017-05-08 00:00:01.869] Notice: Increasing domains struct size from 0 to 1000 (343.45 KB)
[2017-05-08 00:00:01.870] Notice: Increasing clients struct size from 0 to 10 (343.62 KB)
[2017-05-08 00:00:02.021] New client: 127.0.0.1 localhost (1/10)
[2017-05-08 00:00:02.913] Reading from /var/log/pihole.log (rwxrwxr-x)

```

```
root@RPi-Zero-W:~# cat /var/log/syslog | grep 00:00
May  8 00:00:01 RPi-Zero-W rsyslogd-2007: action 'action 17' suspended, next retry is Mon May  8 00:01:01 2017 [try http://www.rsyslog.com/e/2007 ]
May  8 00:00:01 RPi-Zero-W CRON[3283]: (root) CMD (   PATH="$PATH:/usr/local/bin/" pihole flush)
May  8 00:00:01 RPi-Zero-W CRON[3279]: (CRON) info (No MTA installed, discarding output)

```

--------------------

With DietPi-RAMlog mode 2:

[image]

```
root@NanoPi-Neo:~# ls -lah /var/log/ | grep pihole
-rwxrwxr-x  1 pihole   pihole      1 Mai  8 08:17 pihole-FTL.log
-rwxrwxr-x  1 dnsmasq  www-data 1,9M Mai  8 18:55 pihole.log
-rwxrwxr-x  1 root     users       2 Mai  8 00:00 pihole.log.1

root@NanoPi-Neo:~# ls -lah /var/log/ | grep sys
-rwxrwxr-x  1 root     adm         0 Mär 18 18:01 syslog

```

```
root@NanoPi-Neo:~# cat /var/log/pihole-FTL.log

```

```
root@NanoPi-Neo:~# cat /var/log/syslog

```

```
root@NanoPi-Neo:~# cat /var/log/pihole.log.1

```

`pihole-FTL.log` , `pihole.log.1` and `syslog` are touched but empty.
It seems, this will not working without `logrotate`.

```
root@NanoPi-Neo:~# cat /etc/.pihole/advanced/pihole.cron
# Pi-hole: A black hole for Internet advertisements
# (c) 2017 Pi-hole, LLC (https://pi-hole.net)
# Network-wide ad blocking via your own hardware.
#
# Updates ad sources every week
#
# This file is copyright under the latest version of the EUPL.
# Please see LICENSE file for your rights under this license.
#
#
#
# This file is under source-control of the Pi-hole installation and update
# scripts, any changes made to this file will be overwritten when the softare
# is updated or re-installed. Please make any changes to the appropriate crontab
# or other cron file snippets.

# Pi-hole: Update the ad sources once a week on Sunday at 01:59
#          Download any updates from the adlists
59 1    * * 7   root    PATH="$PATH:/usr/local/bin/" pihole updateGravity

# Pi-hole: Update Pi-hole! Uncomment to enable auto update
#30 2    * * 7   root    PATH="$PATH:/usr/local/bin/" pihole updatePihole

# Pi-hole: Flush the log daily at 00:00 so it doesn't get out of control
#          Stats will be viewable in the Web interface thanks to the cron job above
#          The flush script will use logrotate if available
00 00   * * *   root    PATH="$PATH:/usr/local/bin/" pihole flush

@reboot root /usr/sbin/logrotate /etc/pihole/logrotate

```

```
root@NanoPi-Neo:~# cat /etc/pihole/logrotate
/var/log/pihole.log {
        su root root
        daily
        copytruncate
        rotate 5
        compress
        delaycompress
        notifempty
        nomail
}
```

Fourdee, 2017-05-08:
@k-plan 

> With DietPi-RAMlog mode 2

```logrotate``` should be removed in this mode, please can you check:
```
dpkg -l | grep logrotate
```

k-plan, 2017-05-08:
@Fourdee 

tested on different devices, both are `DietPi v.150` and the last PiHole Version, installed via `dietpi-software` :

> With DietPi-RAMlog mode 2

[image]

```
root@NanoPi-Neo:~# dpkg -l | grep logrotate
root@NanoPi-Neo:~#
```

--------------

> With DietPi-RAMlog mode FULL:

[image]

```
root@RPi-Zero-W:~# dpkg -l | grep logrotate
ii  logrotate                     3.8.7-1                   armhf        Log rotation utility

```

k-plan, 2017-05-08:
@Fourdee 

leave it as it is.

Why? 
I personal never  will use buggy shi..y logrotate.

```
root@RPi-Zero-W:~# cat /var/log/syslog | grep 00:00
May  8 00:00:01 RPi-Zero-W rsyslogd-2007: action 'action 17' suspended, next retry is Mon May  8 00:01:01 2017 [try http://www.rsyslog.com/e/2007 ]
```

It's an very old bug and it's easy to fix, but what will happens? Nothing!

https://github.com/Fourdee/DietPi/issues/313#issuecomment-219564276

https://github.com/Fourdee/DietPi/issues/311#issuecomment-230571616

(It is not your fault and you don't have to fix, but the ignorance of some comments like "_it is a bug in the Debian configuration of rsyslogd which is fixed in Stretch_" irritating me, sorry).

And I don't like so fiddling around, in which way third party software works exactly.

WaLLy3K, 2017-05-10:
Sorry, I meant to reply to this sooner. My DietPi install doesn't have `logrotate` (for whatever the reason may be), however [Pi-hole does use `logrotate`](https://github.com/pi-hole/pi-hole/blob/master/advanced/logrotate) to manage its logs.

DL6ER, 2017-05-10:
Not exactly sure how things are right now. Let me summarize how `FTL` works with the files, real quick.

`FTL` glues to `pihole.log` and reads any changes in this file. Things are different with `pihole.log.1`, since this file is *not* expected to change without a connected change in `pihole.log`.

On startup and when the size of `pihole.log` decreases (usually because of some `logrotate` action), the `loganalyzer` starts re-parsing the entire data. This is done to allow for an easy and meanwhile efficient GC algorithm to be used in `FTL`. It means `FTL` *now* (and only in this case) first tries to read `pihole.log.1` and then `pihole.log`. If `pihole.log.1` is not found on the system, that is fine and it is simply skipped.

zeno-bg, 2017-05-18:
Hey all, can anyone say what happens if the tmfps gets full before it gets cleared?

Fourdee, 2017-05-19:
@zeno-bg 

> Hey all, can anyone say what happens if the tmfps gets full before it gets cleared?

I'd assume no new log entries would be created.

However, we limit size of the PiHole log file to 5MB. Checked every hour, if its over this limit, its cleared.

1activegeek, 2018-07-10:
Hey guys, sorry to potentially necro this - but what was the end result and value here around the logging? I only ask as I'm about to open an issue with PiHole team on their repo for something happening to me. I just want to rule out the logging being the issue, but it could potentially be the issue.

Short - The block page is no longer the actual blockage, just a white page saying blocked by pihole, the query log is empty, and the graphs/charts are not updating with anything, BUT - the pages are still being blocked it would seem. This happens after a period of time. If I reboot the device (Rock64) - it comes back up and works as expected. 

I was thinking this could be something to do with the whole 3.3 vs 3.2.1 thing - but current version of DietPi on my Rock64 is Debian Stretch and the Dnsmasq version is the version they call out requiring for the log-queries=extra config.

MichaIng, 2018-07-10:
The white block page is actually an issue of the DietPi implementation, but it means that it actually works at least. Please see and switch to the opened Pi-hole issue, I found some workaround for now, but not yet implemented. We are waiting for next Pi-hole release which brings a less intrusive installation method. I will recheck later, currently just on mobile :wink:.

1activegeek, 2018-07-10:
@MichaIng ya thats the only good thing I took from it! 😉 I was running free and clear from a container on a server, but wanted to switch it off the container for other reasons. I have the R64 running HASS, so I figure it's an easy little add on.

Ok are you referencing #1696? I came upon that, but saw how long it was honestly and kind of wandered away. 😏  Does that cover the logging based issue as well or just the lighttpd part? I'm thinking the core issue I had is actually around this logging thing since it basically "halts" after roughly an hour. And the pihole.log.1 file is 0 bytes all the time. I noticed also that it's owned by root:root vs dnsmasq:www like the pihole.log file.

Fourdee, 2018-07-10:
@1activegeek 

Hi,

Currently, we are waiting for the next release (or pre-RC) of Pi-Hole. 
Once that is out, we can review what needs to be done our end and rewrite/update the installation, hopefully, getting all these outstanding items resolved:
https://github.com/Fourdee/DietPi/issues/1696