MichaIng/DietPi #921: how turn off wifi hotspot?

xor256, 2017-05-04:
I use dietpi-config.
7. adv network
      DietPi - Network Options: Adapters
      Wifi         : Available | Enabled | Wifi Hotspot Mode 
 i go to the 
              │           WiFi     Change Wireless Network Settings           │ 
  and have trouble
[image] 
is impossible to turn of wifi, turn off wifihotspot or i dont know how change my wifi hotspot to ap ot other mode my wifi card

Fourdee, 2017-05-05:
@xor256 
Uninstalling WiFi hotspot in ```dietpi-software``` will return WiFi to standard client mode.

I'll add support for enable/disable WiFi in DietPi-Config: https://github.com/Fourdee/DietPi/issues/924

In the mean time
Turning WiFi hotspot off:
```
systemctl stop hostapd
```

Turning WiFi hotspot back on:
```
systemctl start hostapd
```

> and other small things. I cant setup my region , alvays I get error: Yourchoice is not suported.

We generally have this issue on 3.x kernels that lack support for this feature. Which device are you running?

Fourdee, 2017-05-15:
Marking as completed: https://github.com/Fourdee/DietPi/issues/924

Please reopen if required.

closed 2017-05-15 by Fourdee

FaserF, 2019-01-30:
I am currently getting the same issue. Had WiFi disabled, today I have enabled it so the driver was loaded. Then I used a software which turned my wifi into hotspot. Now I don't need it (and need it to be removed again) but there is no uninstall wifi oder deactivate wifi. Only deactivate hotspot which seems not to work.