MichaIng/DietPi #926: XU4 | Kernel 4.9 testing

Fourdee, 2017-05-05:
>

Fourdee, 2017-05-05:
- Cloudshell Screen
- Cloudshell IR
- DietPi-Cloudshell
- 🈺 Kodi = https://github.com/Fourdee/DietPi/issues/899#issuecomment-299452995 + 
```
wget http://oph.mdrjr.net/meveric/kernel/XU3/4.9.x/xf86-video-armsoc-odroid_20170419-1.4.1-1+deb8_armhf.deb -O package.deb
dpkg -i package.deb
```

nickfox-taterli, 2017-05-06:
I test but not working with waveshare32.

 ───────────────────────────────────────
 DietPi     | 16:44 | Sat 06/05/17
 ───────────────────────────────────────
 V149       | oDroid XU3/4 (armv7l)
 ───────────────────────────────────────
 IP Address | 192.168.99.141
 ───────────────────────────────────────

 Created by : Daniel Knight
 Web        : http://DietPi.com
 Twitter    : http://twitter.com/dietpi_
 Donate     : http://goo.gl/pzISt9

 Device image possible thanks to: Meveric
 DietPi's web hosting is powered by: MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

root@DietPi:~# dmesg | grep fbtft
[    3.401338] fbtft: module is from the staging directory, the quality is unkno                                                                                                                                                             wn, you have been warned.
[    3.410109] fbtft_device: module is from the staging directory, the quality i                                                                                                                                                             s unknown, you have been warned.
[    3.414801] fbtft_device: display not supported: 'odroidc_tft32'
root@DietPi:~# lsmod
Module                  Size  Used by
ads7846                11687  0
spidev                  6446  0
spi_s3c64xx            11643  0
exynos_gpiomem          3162  0
fbtft                  24420  0
fuse                   75590  1
root@DietPi:~# modprobe fbtft_device
modprobe: ERROR: could not insert 'fbtft_device': Invalid argument
root@DietPi:~#

[    3.401338] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    3.410109] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[    3.414801] fbtft_device: display not supported: 'odroidc_tft32'
[ 8571.621657] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[ 8571.622475] fbtft_device: display not supported: 'odroidc_tft32'

Fourdee, 2017-05-07:
@nickfox-taterli 

Thanks.

Did you enable Cloudshell/waveshare32 through ```dietpi-config``` > Display options > LCD panel?

Fourdee, 2017-05-09:
Test possible fix for vsync tearing in Kodi video playback (thanks to Meveric):
advancedsettings.xml
```
<advancedsettings>
    <video>
        <adjustrefreshrate>
            <override>
                <fpsmin>23</fpsmin>
                <fpsmax>60</fpsmax>
                <refresh>60</refresh>
            </override>
        </adjustrefreshrate>
    </video>
</advancedsettings>
```

justin0a0, 2017-05-10:
Tried testing with the image you posted, on first login I get:
[Failed] Insufficient free space on RootFS. 30 MB available, 500 MB required
[Failed] Aborting

I assume this is from you checking for free space before the updates but seeing as the partition was not yet resized I got this error.

Fourdee, 2017-05-10:
@justin0a0 

> I assume this is from you checking for free space before the updates but seeing as the partition was not yet resized I got this error.

Thanks for report 👍 
Might be an issue with mmcblk index's, was this on SD or EMMC? And any chance you can run/paste results:
```
blkid
```

nickfox-taterli, 2017-05-10:
@Fourdee only cloudshell (hktft9340) install correct,i guess it miss " .controller_data = &hktft9340_controller_data, " at fbtft_device.c.

justin0a0, 2017-05-13:
@Fourdee 
It was while using an SD I have been trying to find time to put the SD back in but the house gets a bit upset when they can't access the media I'll give it another run this weekend if you still need the info

joaofl, 2017-05-13:
@Fourdee 
I've burned the testing image, which booted fine. But after running dietpi's tool to transfer the rootfs to the usb, the boot.ini file ended getting an the wrong UUID, which I had to set manually in order to boot. The one on fstab was correct.

Fourdee, 2017-05-13:
@joaofl 

Thanks for report 👍  I'll take a look.

joaofl, 2017-05-22:
@Fourdee 

> Strange, i've checked the code (v150), same UUID var is applied to fstab and boot.ini. I'll try to replicate on XU4.

Maybe I messed it up, or maybe because I had 2 USB drives plugged in? I'll try to replicate it and I let you know if anything happens.

joaofl, 2017-05-23:
@Fourdee 
I actually reproduced the issue I was having, but it ended up not being due to the UUID  I think. However I've observed a few things.

(1) After transferring the root partition to an external drive and rebooting, dietpi starts to auto-mount (line added to fstab) the old root partition of the sd card as an external drive. Something we actually don't want, right?

(2) When we transfer the root, you update the UUID in a couple of files, right? My idea was: Can't you update this files only on the new transfered rootfs, on the USB drive only (and not before rsync)? Because this way, if we want to go back to the previous scheme, we only need to modify the boot.ini file on the boot partition (maybe by uncommenting a line). This would allow for easy rollback in case the transfer fails. 

(3) The problem I was having was actually due the spin-up delay of one of my external HDs on a cold boot. It means that if I simply reboot the system, it works fine. But the delay of spinning-up the disk before mouting it on a cold boot, ends up blocking the booting, even after successfully mounting the HD. Something like a deadlock. Have an idea on how to solve this?

Everything else seems fine so far with the 4.9.24 kernel. (4.9.28 is available by meveric, should I try to update?)

joaofl, 2017-05-23:
@Fourdee 
I actually just checked your code @ https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-drive_manager and it actually already implements (2) and something regarding (1) as well. So I'm not sure... By the way, I'm running dietpi v150.

Fourdee, 2017-05-23:
@joaofl 

> and it actually already implements (2) 

Yep:
if Rsync fails with rootfs transfer, the original ```/etc/fstab``` is restored, then script exited to prevent further changes to ```boot.ini``` etc.:
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-drive_manager#L497-L507

> The problem I was having was actually due the spin-up delay of one of my external HDs on a cold boot. It means that if I simply reboot the system, it works fine.

On the rpi, adding ```rootdelay=10``` to ```cmdline.txt``` helps resolve this. Try doing ```rootdelay=10``` to ```boot.ini```, in line ```setenv bootrootfs ""```

> After transferring the root partition to an external drive and rebooting, dietpi starts to auto-mount (line added to fstab) the old root partition of the sd card as an external drive. Something we actually don't want, right?

The existing entry should be removed. Any chance you can paste results of following after this occurs?
```
blkid
cat /etc/fstab
cat /DietPi/boot.ini
```

joaofl, 2017-05-23:
@Fourdee 

> Try doing rootdelay=10 to boot.ini, in line setenv bootrootfs ""

It worked. Thanks!

> the original /etc/fstab is restored, then script exited to prevent further changes to boot.ini

Thats true. But it did not work for me, because I had to interrupt the rsync with ctrl+c. This because it blocked during the copy, most probably because of some badblocks on the sd card. So maybe, if this part of the code comes after the rsync.... But anyway, thats not a big deal.

4.9.28 is available by meveric, should the update happen seamlessly?

> Any chance you can paste results of following after this occurs?

blkid
```
/dev/mmcblk1: PTUUID="000f1766" PTTYPE="dos"
/dev/mmcblk1p1: SEC_TYPE="msdos" LABEL="boot" UUID="5886-8548" TYPE="vfat" PARTUUID="000f1766-01"
/dev/mmcblk1p2: LABEL="rootfs" UUID="a4f12232-1670-43f4-8eca-49bfbe4d01ee" TYPE="ext4" PARTUUID="000f1766-02"
/dev/sda1: LABEL="DATA" UUID="23504cb3-a823-422f-a1fa-61227456189c" TYPE="ext4" PARTLABEL="My Book" PARTUUID="69cd0e21-955d-4c89-925f-495a5395beb3"
/dev/sdb1: UUID="540cd7e8-4d07-4d36-9eb5-1fa8a31f534e" TYPE="ext4" PARTUUID="90991f47-01"

```

fstab
```
#Internal Drives---------------------------------------------------
proc            /proc           proc    defaults				0 0
UUID=5886-8548  /boot           auto    defaults,noatime,discard		0 2
UUID=540cd7e8-4d07-4d36-9eb5-1fa8a31f534e / auto defaults,noatime,discard 0 1
tmpfs 			/tmp  			tmpfs 	defaults,noatime,nodev,nosuid,mode=1777				0 0
tmpfs 			/var/log 		tmpfs 	defaults,size=20m,noatime,nodev,nosuid,mode=1777	0 0
tmpfs 			/DietPi 		tmpfs 	defaults,size=10m,noatime,nodev,nosuid,mode=1777	0 0

#External Drives---------------------------------------------------
#NB: Please use dietpi-drive_manager to setup and control your external drives.

#Samba Client------------------------------------------------------
#/mnt/samba . Please use dietpi-config and the Networking Options: NAS menu to setup this mount

#FTP Client Mount--------------------------------------------------
#/mnt/ftp_client . Please use dietpi-config and the Networking Options: NAS menu to setup this mount

#NFS Client Mount--------------------------------------------------
#/mnt/nfs_client . Please use dietpi-config and the Networking Options: NAS menu to setup this mount
UUID=23504cb3-a823-422f-a1fa-61227456189c       /mnt/hd      auto     defaults,noatime,nofail,x-systemd.automount  0 0
#UUID=a4f12232-1670-43f4-8eca-49bfbe4d01ee       /mnt/a4f12232-1670-43f4-8eca-49bfbe4d01ee      auto     defaults,noatime,nofail,x-systemd.automount  0 0
```

boot.ini
```
#------------------------------------------------------------------------------------------------------
# Basic Ubuntu Setup. Don't touch unless you know what you are doing.
# --------------------------------
setenv bootrootfs "console=tty1 root=UUID=540cd7e8-4d07-4d36-9eb5-1fa8a31f534e rootwait ro fsck.repair=yes rootdelay=20"

# boot commands
setenv bootcmd "fatload mmc 0:1 0x40008000 zImage; fatload mmc 0:1 0x42000000 uInitrd; fatload mmc 0:1 0x44000000 exynos5422-odroidxu4.dtb; bootz 0x40008000 0x42000000 0x44000000"
```

Fourdee, 2017-05-24:
@joaofl 

> Thats true. But it did not work for me, because I had to interrupt the rsync with ctrl+c. This because it blocked during the copy, most probably because of some badblocks on the sd card. So maybe, if this part of the code comes after the rsync.... But anyway, thats not a big deal.

Ah yes, I see, not much we can do about ctrl+c. Will require a manual mv of the backup fstab.

> 4.9.28 is available by meveric, should the update happen seamlessly?

For now, you'll need to install/update these manually. Meveric is due to update his repo, replacing the current linux packages to 4.9. Once done, ```apt-get upgrade``` would do it.

Fourdee, 2017-05-30:
Meveric has provided apt packages for Kodi 17, needs testing:
https://forum.odroid.com/viewtopic.php?p=191780#p191780

Fourdee, 2017-06-06:
Meveric is updating the 3.10 kernel to support EMMC 5.1. To avoid this kernel overwriting the 4.9 kernel, users will need to put kernel packages on hold:
```
apt-mark hold linux-image-armhf-odroid-xu3
```

Fourdee, 2017-06-08:
> After upgrading to 4.9 kernel the IR-remote didn't work anymore. I had to rerun the ./install_ir.sh script to fix the problem.
http://dietpi.com/phpbb/viewtopic.php?f=11&t=1937&p=7466#p7407

joaofl, 2017-06-08:
@Fourdee 
I've been running the 4.9.29 version for some days already, and had no major problems. It is quiet stable, and the performance upgrade on the ethernet is  visible.

But I keep getting the following warning:
`usb 4-1.1: reset SuperSpeed USB device number 3 using xhci-hcd`

It happens specially when, from a remote machine, I upload and download files (simultaneously) from/to the hard-drive mounted on my XU4 USB3 on the network.

The problem does not seems to be related to the USB3 itself, since when working locally, I can read and write files from/to the hard drive with no warnings at all.

So, it seems to be related somehow to Ethernet and USB operations happening simultaneously.

I've searched around for similar issues. A guy had similar problem, that got solved but he doesn't know how.

Do you have any clue what it can be or where to look for answers?
Thanks.

Fourdee, 2017-06-12:
@joaofl 

> I've been running the 4.9.29 version for some days already, and had no major problems. It is quiet stable, and the performance upgrade on the ethernet is visible.

Yep same here, running 24/7 NAS + PiHole. Runs great.

> But I keep getting the following warning:
usb 4-1.1: reset SuperSpeed USB device number 3 using xhci-hcd

Could be a compatibility issue with USB host/adapter/kernel. 

> It happens specially when, from a remote machine, I upload and download files (simultaneously) from/to the hard-drive mounted on my XU4 USB3 on the network.

Might be power related if USB drive is host powered (lack of under load), which PSU are you running ?

joaofl, 2017-06-12:
@Fourdee 
> Might be power related if USB drive is host powered (lack of under load), which PSU are you running ?
The hard drive is externally powered, and I'm not actually draining much current from the XU4 at all, and the power supply I use can supply +10A, so It should not be an issue. And it never happened with the kernel version 3.

As you said, It may be a compatibility problem? Not sure how to debug that....

This is the HD I'm currently using:
https://www.wdc.com/products/wd-recertified/my-book.html

Fourdee, 2017-06-12:
@joaofl 

> RECERTIFIED

Could also be an outstanding issue with the unit, that didn't get fixed (eg: USB > SATA controller).

joaofl, 2017-06-14:
@Fourdee 
The closest I found was this thread:
http://linux-scsi.vger.kernel.narkive.com/B3jfin3j/xhci-issues-wd-mybook-1230-reset-superspeed-usb-device
Which seems to be kind of a deeper issue, on the device driver itself... Not sure what to do. There is no firmware upgrades for the HD so far.

filemoon, 2017-06-14:
Kernel 4.9 (29+)

Only performance governor is present.
Do you have a plan to add other governors (I'm looking at on-demnd and conservative)

Regards Chris

Fourdee, 2017-06-14:
@filemoon 

Hi Chris, please see this post for more info:
http://dietpi.com/phpbb/viewtopic.php?f=9&t=1927&p=7388&hilit=ondemand#p7388

Fourdee, 2017-06-16:
- Need to update our 4.9 testing image due to https://github.com/Fourdee/DietPi/issues/926#issuecomment-306546305 which is automatically installed during dietpi-software first run.

Fourdee, 2017-06-25:
Update from Meveric, testing and feedback required:
> today I've uploaded a new version of the Kernel for 4.9 with updated meta packages. 
They will remove all currently installed 3.10 Kernels (but keep the headers, since they don't really matter).
Packages are as mentioned before: ```linux-image-4.9-armhf-odroid-xu3``` and ```linux-headers-4.9-armhf-odroid-xu3```
It would be nice if you or some of your community members could try it and give a little feedback how it works.
If you do an update from 3.10.XXX to 4.9 it will complain that it may be insecure to remove the currently loaded Kernel. 
It will ask if it should STOP the removal.
So the correct answer here is NO.
Looking forward to some feedback.

filemoon, 2017-06-26:
Freshly noticed issue:

HMP not active for ARM big.LITTLE (only big cores are in use):
- only 4 (big) cores available (another 4 small cores not in use)
- high temperature
- slower performance (vs 8 cores)
- high power consumption

I will restore the backup and wait for another kernel 4.9 build with HMP activated and more governors.

studioai, 2017-07-04:
http://odroid.com/dokuwiki/doku.php?id=en:xu4_upgrade_from_3.10_to_4.9

Hardkernel has updated their wiki page and repository.

Fourdee, 2017-07-04:
@studioai 

> http://odroid.com/dokuwiki/doku.php?id=en:xu4_upgrade_from_3.10_to_4.9
Hardkernel has updated their wiki page and repository.

This does not apply to our image as we are using Meveric's repo, which contains 4.9 kernel packages:
```
https://github.com/Fourdee/DietPi/issues/926#issuecomment-310918133
```

rncwnd79, 2017-07-04:
> Needs testing on 3.x image (as upgrade)

I tried Meverics exact steps @Fourdee and get this (on 3.x image, already running manually installed 4.9):
`Fehl http://fuzon.co.uk/meveric/ all/xu3 linux-image-4.9-armhf-odroid-xu3 armhf 4.9.34-1
  404  Not Found`
`E: Fehlschlag beim Holen von http://fuzon.co.uk/meveric/pool/xu3/l/linux-image-4.9-armhf-odroid-xu3/linux-image-4.9-armhf-odroid-xu3_4.9.34-1_armhf.deb  404  Not Found`

`E: Einige Archive konnten nicht heruntergeladen werden; vielleicht »apt-get update« ausführen oder mit »--fix-missing« probieren?`

joaofl, 2017-07-04:
@Fourdee 
Just to understand, whats the difference between meverics compile and others?

I keep track of his releases, which was last updated end of may (v4.9.29), while the odroid repo is already at version 45?
(https://oph.mdrjr.net/meveric/kernel/XU3/4.9.x/)
(https://github.com/hardkernel/linux/tree/odroidxu4-4.9.y)

Fourdee, 2017-07-04:
@rncwnd79 

> E: Fehlschlag beim Holen von http://fuzon.co.uk/meveric/pool/xu3/l/linux-image-4.9-armhf-odroid-xu3/linux-image-4.9-armhf-odroid-xu3_4.9.34-1_armhf.deb 404 Not Found

Confirmed earlier today.

Its functional now, please re-run:
```
apt-get update
apt-get install linux-image-4.9-armhf-odroid-xu3
```

Most likely Meveric was syncing our mirror server and its only just completed (fuzon.co.uk)

```
root@DietPi:~# apt-get install linux-image-4.9-armhf-odroid-xu3
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libevent-2.0-5 libhunspell-1.3-0 linux-image-4.9.33+
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  linux-image-4.9.35+
Suggested packages:
  fdutils linux-doc-4.9.35+ linux-source-4.9.35+ linux-image-4.9.35+-dbg
  linux-manual-4.9.35+
Recommended packages:
  kernel-common
The following NEW packages will be installed:
  linux-image-4.9.35+
The following packages will be upgraded:
  linux-image-4.9-armhf-odroid-xu3
1 upgraded, 1 newly installed, 0 to remove and 22 not upgraded.
Need to get 14.2 MB of archives.
After this operation, 44.6 MB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://fuzon.co.uk/meveric/ all/xu3 linux-image-4.9.35+ armhf 4.9.35-20170704-XU3 [14.2 MB]
Get:2 http://fuzon.co.uk/meveric/ all/xu3 linux-image-4.9-armhf-odroid-xu3 armhf 4.9.35-1 [1,378 B]
Fetched 14.2 MB in 3s (4,472 kB/s)
Preconfiguring packages ...
Selecting previously unselected package linux-image-4.9.35+.
(Reading database ... 36834 files and directories currently installed.)
Preparing to unpack .../linux-image-4.9.35+_4.9.35-20170704-XU3_armhf.deb ...
Examining /etc/kernel/preinst.d/
run-parts: executing /etc/kernel/preinst.d/remove-old-on-upgrade 4.9.35+ /boot/vmlinuz-4.9.35+
Done.
Unpacking linux-image-4.9.35+ (4.9.35-20170704-XU3) ...
Replacing files in old package linux-image-4.9.33+ (4.9.33-20170620-XU3) ...
Preparing to unpack .../linux-image-4.9-armhf-odroid-xu3_4.9.35-1_armhf.deb ...
Unpacking linux-image-4.9-armhf-odroid-xu3 (4.9.35-1) over (4.9.33-1) ...
Setting up linux-image-4.9.35+ (4.9.35-20170704-XU3) ...

 Hmm. There is a symbolic link /lib/modules/4.9.35+/build
 However, I can not read it: No such file or directory
 Therefore, I am deleting /lib/modules/4.9.35+/build

 Hmm. The package shipped with a symbolic link /lib/modules/4.9.35+/source
 However, I can not read the target: No such file or directory
 Therefore, I am deleting /lib/modules/4.9.35+/source

Running depmod.
Examining /etc/kernel/postinst.d.
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 4.9.35+ /boot/vmlinuz-4.9.35+
run-parts: executing /etc/kernel/postinst.d/copy_dtb 4.9.35+ /boot/vmlinuz-4.9.35+
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 4.9.35+ /boot/vmlinuz-4.9.35+
update-initramfs: Generating /boot/initrd.img-4.9.35+
Image Name:   uInitrd
Created:      Tue Jul  4 17:03:05 2017
Image Type:   ARM Linux RAMDisk Image (uncompressed)
Data Size:    4186917 Bytes = 4088.79 kB = 3.99 MB
Load Address: 00000000
Entry Point:  00000000
run-parts: executing /etc/kernel/postinst.d/uInitrd 4.9.35+ /boot/vmlinuz-4.9.35+
run-parts: executing /etc/kernel/postinst.d/zImage 4.9.35+ /boot/vmlinuz-4.9.35+
Setting up linux-image-4.9-armhf-odroid-xu3 (4.9.35-1) ...
```

Fourdee, 2017-07-04:
@joaofl 

> I keep track of his releases, which was last updated end of may (v4.9.29), while the odroid repo is already at version 45?

I believe the x.x.XX code reflects the build count. So each time Meveric updates his kernel with latest, his build count number will be unique to how many times hes done it.

I could be wrong, but you'd need to ask Meveric on Odroid forums to confirm.

> Just to understand, whats the difference between meverics compile and others?

As far as i'am aware, they are indentical, minus any tweaks by Meveric (eg: module options), but again, you'd need to confirm with Meveric.

Its worth noting, Meveric's repo and his continued maintenance of it, allows for everyone to easily update their kernel via APT, without compiling each time its updated.

studioai, 2017-07-05:
@Fourdee 

> Its functional now, please re-run:
> 
> apt-get update
> apt-get install linux-image-4.9-armhf-odroid-xu3

I tried those command and I got 

```
root@DietPi:~# apt-get install linux-image-4.9-armhf-odroid-xu3
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-image-4.9-armhf-odroid-xu3
E: Couldn't find any package by regex 'linux-image-4.9-armhf-odroid-xu3'
```

Do I need to edit source.list?

Fourdee, 2017-07-05:
@studioai 

Strange, do you have all the lists (mainly the XU3 one)?
```
root@DietPi:~# ls -lha /etc/apt/sources.list.d/
total 24K
drwxr-xr-x 2 root root 4.0K Jun 20 11:44 .
drwxr-xr-x 6 root root 4.0K Jun 20 11:44 ..
-rw-r--r-- 1 root root   48 May 19 13:04 meveric-all.list
-rw-r--r-- 1 root root   39 May 19 13:04 meveric-all-XU3.list
-rw-r--r-- 1 root root   53 May 19 13:04 meveric-jessie.list
```

```
root@DietPi:~# apt-cache showpkg linux-image-4.9-armhf-odroid-xu3
Package: linux-image-4.9-armhf-odroid-xu3
Versions:
4.9.35-1 (/var/lib/apt/lists/fuzon.co.uk_meveric_dists_all_xu3_binary-armhf_Packages) (/var/lib/dpkg/status)
 Description Language:
                 File: /var/lib/apt/lists/fuzon.co.uk_meveric_dists_all_xu3_binary-armhf_Packages
                  MD5: 9c9c3754dda52eaadfc61ec69d5a7d6d
```

And ```apt-get update``` ran before hand?

studioai, 2017-07-05:
@Fourdee 

Nope.  There's only meveric-jessie.list.

```
root@DietPi:~# ls -lha /etc/apt/sources.list.d/
total 5.0K
drwxr-xr-x 2 root root 1.0K Mar  5 23:15 .
drwxr-xr-x 6 root root 1.0K Mar  5 23:15 ..
-rw-r--r-- 1 root root   55 Mar  5 23:15 deb-multimedia.list
-rw-r--r-- 1 root root   53 Mar  5 23:15 meveric-jessie.list
-rw-r--r-- 1 root root   40 Mar  5 23:15 plex.list
```

I didn't touched lists at all since I installed dietpi last year. 

Anyway I'll try again after downloading those lists.

studioai, 2017-07-05:
@Fourdee 

Adding meveric-all-XU3.list to /etc/apt/sources.list.d/ solved my problem. Thank you!

rncwnd79, 2017-07-05:
Trying right now.

Btw.
`--fix-missing`
worked after cleaning /boot.

studioai, 2017-07-07:
OnDemand Governor seems to does not work properly. 

[image]
[image]
[image]

CPU4~8 Frequency fixed on 2000MHz and never go down.

Fourdee, 2017-07-07:
@studioai 

> CPU4~8 Frequency fixed on 2000MHz and never go down.

Our CPU script will tend to cause the ondemand throttle up to trigger. There is nothing we can do about this inside bash.

Try:
```
cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_cur_freq
```

Results in:
```
root@DietPi:~# cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_cur_freq
200000
200000
200000
200000
700000
700000
700000
700000
```

studioai, 2017-07-08:
@Fourdee 
`cat /sys/devices/system/cpu/cpu*/cpufreq/stats/time_in_state`

That gives me reasonable results. look good.
Thanks!

filemoon, 2017-07-24:
@Fourdee

Do we have any update about HMP support in 4.9 (XU4)?

Odroid says this should be working but when I've checked your kernel this is not working.
What I see:
- only big cores are occupied
- sometimes I can see small task on small core number 1 (only number 1)
- core number 5 is always most utilised core
- next in line is 6 .... etc

[image]
[image]
[image]
[image]

studioai, 2017-07-25:
updating kernel to 4.9.37 from 4.9.35 is unavailable. 

```
root@DietPi:~# apt-get install linux-image-4.9-armhf-odroid-xu3 linux-headers-4.9-armhf-odroid
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-4.9.35+ linux-image-4.9.35+
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  linux-headers-4.9.37+ linux-image-4.9.37+
Suggested packages:
  module-init-tools fdutils linux-doc-4.9.37+ linux-source-4.9.37+ linux-image-4.9.37+-dbg lin
Recommended packages:
  kernel-common
The following NEW packages will be installed:
  linux-headers-4.9.37+ linux-image-4.9.37+
The following packages will be upgraded:
  linux-headers-4.9-armhf-odroid-xu3 linux-image-4.9-armhf-odroid-xu3
2 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 22.3 MB of archives.
After this operation, 103 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://fuzon.co.uk/meveric/ all/xu3 linux-image-4.9.37+ armhf 4.9.37-20170724-XU3 [14.2 
Get:2 http://fuzon.co.uk/meveric/ all/xu3 linux-headers-4.9.37+ armhf 4.9.37-20170724-XU3 [8,0
Get:3 http://fuzon.co.uk/meveric/ all/xu3 linux-headers-4.9-armhf-odroid-xu3 armhf 4.9.37-1 [9
Get:4 http://fuzon.co.uk/meveric/ all/xu3 linux-image-4.9-armhf-odroid-xu3 armhf 4.9.37-1 [1,3
Fetched 22.3 MB in 16s (1,335 kB/s)                                                           
Preconfiguring packages ...
Selecting previously unselected package linux-image-4.9.37+.
(Reading database ... 48704 files and directories currently installed.)
Preparing to unpack .../linux-image-4.9.37+_4.9.37-20170724-XU3_armhf.deb ...
Examining /etc/kernel/preinst.d/
run-parts: executing /etc/kernel/preinst.d/remove-old-on-upgrade 4.9.37+ /boot/vmlinuz-4.9.37+
Done.
Unpacking linux-image-4.9.37+ (4.9.37-20170724-XU3) ...
Replacing files in old package linux-image-4.9.35+ (4.9.35-20170704-XU3) ...
Selecting previously unselected package linux-headers-4.9.37+.
Preparing to unpack .../linux-headers-4.9.37+_4.9.37-20170724-XU3_armhf.deb ...
Unpacking linux-headers-4.9.37+ (4.9.37-20170724-XU3) ...
Preparing to unpack .../linux-headers-4.9-armhf-odroid-xu3_4.9.37-1_armhf.deb ...
Unpacking linux-headers-4.9-armhf-odroid-xu3 (4.9.37-1) over (4.9.35-1) ...
Preparing to unpack .../linux-image-4.9-armhf-odroid-xu3_4.9.37-1_armhf.deb ...
Unpacking linux-image-4.9-armhf-odroid-xu3 (4.9.37-1) over (4.9.35-1) ...
Setting up linux-image-4.9.37+ (4.9.37-20170724-XU3) ...

 Hmm. There is a symbolic link /lib/modules/4.9.37+/build
 However, I can not read it: No such file or directory
 Therefore, I am deleting /lib/modules/4.9.37+/build

 Hmm. The package shipped with a symbolic link /lib/modules/4.9.37+/source
 However, I can not read the target: No such file or directory
 Therefore, I am deleting /lib/modules/4.9.37+/source

Running depmod.
Examining /etc/kernel/postinst.d.
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 4.9.37+ /boot/vmlinuz-4.9.37+
run-parts: executing /etc/kernel/postinst.d/copy_dtb 4.9.37+ /boot/vmlinuz-4.9.37+
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 4.9.37+ /boot/vmlinuz-4.9.37+
update-initramfs: Generating /boot/initrd.img-4.9.37+

Socket error Event: 32 Error: 10053.
Connection closing...Socket close.

Connection closed by foreign host.

Disconnected from remote host(studioai.freeddns.org) at 23:52:16.

```

What does mean 'Socket error Event: 32 Error: 10053'? Any information will be appreciated.

Fourdee, 2017-07-25:
@studioai 

There appears to be an issue with the current package, although, mines a completely different error:
```
Running depmod.
Examining /etc/kernel/postinst.d.
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 4.9.37+ /boot/vmlinuz-4.9.37+
run-parts: executing /etc/kernel/postinst.d/copy_dtb 4.9.37+ /boot/vmlinuz-4.9.37+
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 4.9.37+ /boot/vmlinuz-4.9.37+
update-initramfs: Generating /boot/initrd.img-4.9.37+
Image Name:   uInitrd
Created:      Tue Jul 25 17:19:55 2017
Image Type:   ARM Linux RAMDisk Image (uncompressed)
Data Size:    4186621 Bytes = 4088.50 kB = 3.99 MB
Load Address: 00000000
Entry Point:  00000000
run-parts: executing /etc/kernel/postinst.d/uInitrd 4.9.37+ /boot/vmlinuz-4.9.37+
mkimage: Write error on /boot/uInitrd-4.9.37+: Success
run-parts: /etc/kernel/postinst.d/uInitrd exited with return code 1
Failed to process /etc/kernel/postinst.d at /var/lib/dpkg/info/linux-image-4.9.37+.postinst line 391.
dpkg: error processing package linux-image-4.9.37+ (--configure):
 subprocess installed post-installation script returned error exit status 2
Setting up libisc95 (1:9.9.5.dfsg-9+deb8u13) ...
Setting up libdns100 (1:9.9.5.dfsg-9+deb8u13) ...
Setting up libisccc90 (1:9.9.5.dfsg-9+deb8u13) ...
Setting up libisccfg90 (1:9.9.5.dfsg-9+deb8u13) ...
Setting up libbind9-90 (1:9.9.5.dfsg-9+deb8u13) ...
Setting up liblwres90 (1:9.9.5.dfsg-9+deb8u13) ...
Setting up bind9-host (1:9.9.5.dfsg-9+deb8u13) ...
Setting up dnsutils (1:9.9.5.dfsg-9+deb8u13) ...
Setting up locales (2.19-18+deb8u10) ...
Generating locales (this might take a while)...
  en_GB.UTF-8... done
Generation complete.
Setting up lxterminal (0.2.0-1+deb8u1) ...
update-alternatives: using /usr/bin/lxterminal to provide /usr/bin/x-terminal-emulator (x-terminal-emulator) in auto mode
Setting up sudo (1.8.10p3-1+deb8u4) ...
Setting up xarchiver (1:0.5.4-1+deb8u2) ...
Setting up xserver-common (2:1.16.4-1+deb8u1) ...
Setting up xserver-xorg-core (2:1.16.4-1+deb8u1) ...
dpkg: dependency problems prevent configuration of linux-image-4.9-armhf-odroid-xu3:
 linux-image-4.9-armhf-odroid-xu3 depends on linux-image-4.9.37+; however:
  Package linux-image-4.9.37+ is not configured yet.

dpkg: error processing package linux-image-4.9-armhf-odroid-xu3 (--configure):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.19-18+deb8u10) ...
Errors were encountered while processing:
 linux-image-4.9.37+
 linux-image-4.9-armhf-odroid-xu3
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Your issue may be a "one off" loss of connection, or, issues with DNS. Either way, the current package is broken, i'll let meveric know.

Fourdee, 2017-07-26:
Tried again today from a backup, 4.9 kernel updates successfully.

studioai, 2017-08-11:
@Fourdee 
Just FYI

I have met 2 issue while I do upgrading kernel. 

1. Sudden Socket Error and Disconnection
2. I/O Error while update-initramfs

The reason of first one I presume is lack of memory, I set up the swap memory and the issue is gone.
The second one is still under instigation, but I solve the issue with updating kernel after unmounting /boot. When upgrade finish, I copied contents of /boot to /tmp and remount /boot again and overwrite with files that I copied to /tmp. but the reason why I/O Error occurred on /boot is still unknown.

filemoon, 2017-08-11:
/O Error occurred on /boot  - for my case this was no free memory in boot "partition" (space was reclaimed by manually deleting old items)

rncwnd79, 2017-10-20:
I can confirm this.
Issue 2 appears when there is not enough space on /boot..

1989gironimo, 2017-11-15:
My experiences so far:
- If I install the 4.9 image the XU4 stops randomly because of heat issues - the fan doesn't get triggered often enough to cool down the device (don't have this issue if I'm upgrading from 3.x and I can reproduce this behavior) - I cannot get past the installation routine because of this.
- If you upgrade from 3.x you have to change the firmware entry \boot\boot.ini -> exynos-blablaxu from **3** to **4**.dtb) otherwise network won't work
- If you transfer the RootFS to another harddrive (cloudshell 2 with harddrives installed for instance) it's randomly crashing. The ssh-connection still works, but no other application.

`apt-get remove slapd
-bash: /usr/bin/apt-get: Eingabe-/Ausgabefehler
root@randomhostname:/tmp/install_script# nano /var/log/yunohost-installation.log 
-bash: /usr/bin/nano: Eingabe-/Ausgabefehler
root@randomhostname:/tmp/install_script# 
`
translate to "input/output error" 
It seems that the XU4 looses connection to those drives. Again, this doesn't happen on 3.x kernel.

- If you copy the image via dd onto the sd-card, the resizing of the rootfs takes sometimes longer than you're able to connect to it and start the installation. It stops because of space issues, if you wait a little longer, it's working flawlessly.

What can I provide to get this issues fixed? Am I the only one having the issue with the harddrives? Because this is really pissing me off and want to help (even if I'm not a dev)

joaofl, 2017-11-15:
The issue with the HD is there. It happens to me as well, and to some other people, as reported at the hardkernel forums. 

This same issue manifests with the rootfs on the USB, causing  the problem you mentioned (that crashes everything).

In my case, I'm went back to the 3.10 kernel version, which is still more stable (but don't take the best out of the hardware).

The kernel  developers believe that this issue is fixed in kernel 4.14, to be released in the next couple of weeks. Although it make take.longer to arrive on the Dietpi. It may be worth testing the new kernel using the Armbian distro.

secpri, 2017-11-19:
The **(NR_4.x)_DietPi_OdroidXU4-armv7-(Jessie).7z** works so far seemly out of the box on an **ODROID HC-1**, very nice.

[image]

[image]

[image]

1989gironimo, 2017-11-20:
odroid released the final of kernel 4.14, but we have to wait until meveric updates his repository. Am I right?

> Update! 13-Nov-2017
> Kernel 4.14 final has been released.

[https://forum.odroid.com/viewtopic.php?f=146&t=27548](discussion in the odroid-forums)
[https://forum.odroid.com/viewtopic.php?f=146&t=28895](and here's the release information)

Really tempted to try the new version because of the hdd-fan not working on 3.1xx

Fourdee, 2017-11-22:
@1989gironimo 

Meveric update 4.9 kernel recently:

http://fuzon.co.uk/meveric/pool/xu3/l/linux-image-4.9-armhf-odroid-xu3/
```
linux-image-4.9-armhf-odroid-xu3_4.9.61-1_armhf.deb | 2017-11-18 11:09 | 1.4K |  
```

---

Odroid are classing this kernel as testing ```Your system can be corrupted or malfunction with this bleeding edge kernel.```. Not sure why they are focusing on 4.14, when 4.9 seems stable?
https://odroidinc.com/blogs/news/odroid-xu4-kernel-4-9-ubuntu-image-for-public-release

joaofl, 2017-11-22:
@Fourdee
Many people on their forum are reporting this issue with the USB3, which was said to be fixed on the 4.14.
I could not move to 4.9 due to these IO errors, which are really serious, and completely crash the system when root is on the USB 

It also fixes issues with Kodi

1989gironimo, 2017-11-23:
@Fourdee 
I've used exactly that kernel - this behavior isn't fixed on 4.9, even the newest release of meverics kernel.
4.14 is the way to go, see:
[linklink](https://forum.odroid.com/viewtopic.php?f=146&t=27548&start=100)

Fourdee, 2018-02-22:
Marking as closed, due to:

> Please note we will not support Kernel 4.9 anymore due to our limited resoruces. We will keep updating Kernel 4.14 LTS only.
https://forum.odroid.com/viewtopic.php?f=146&t=28895

closed 2018-02-22 by Fourdee