MichaIng/DietPi #927: Request: External USB backup cannot be FAT32 / VFAT

Invictaz, 2017-05-05:
Bit strange that I have to have a EXT formatted USB drive to store backups, otherwise symbolic links are not supported. I want a stationary backup for example in a .zip or something. 

The only thing I can do know is format my drive and then store a backup. But I don't want symbolic links, only a store of my userdata in case Dietpi goes wrong.

Fourdee, 2017-05-06:
> have to have a EXT formatted USB drive to store backups

As the system we backup uses symlinks, EXT4 + symlink filesystem are required for ```dietpi-backup```. If FAT was allowed, a restore would cripple your installation.

If you want to simply backup your userdata to another location, try ```dietpi-sync```: 
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=30#p256

> I want a stationary backup for example in a .zip or something.

https://github.com/Fourdee/DietPi/issues/816

Invictaz, 2017-05-06:
Thanks. Will wait for #816

closed 2017-05-06 by Fourdee