MichaIng/DietPi #928: Faster update if installing package

Invictaz, 2017-05-05:
Situation: v149 is installed.
Installing SABNZBd via dietpi-software

It starts to update and upgrade APT

It upgrades literally every package it finds 
For example gnupg2 2.0.26-6 to 2.0.26-6+deb8u1

And that for hundreds and hundreds of packages. Why?

It's already a half hour underway and no idea when it will be finished.

Why not only major versions or only the packages that are truly necessary for SABNZBd to install?

Fourdee, 2017-05-06:
> It starts to update and upgrade APT

@Invictaz 

This is by design. 
When you install any software in DietPi-Software, APT is updated and upgraded. This ensures all core system software is upto date.

If we dont do this, and for example you install Kodi, which lets say requires  ```gnupg2 2.0.26-6+deb8u1```, it would fail to install without the upgrade beforehand.

> It's already a half hour underway and no idea when it will be finished.

A UHS-3 SD card will vastly improve filesystem IO performance. Example:
- Sandisk Extreme, 1-2 minutes to install Kodi
- Sandisk Ultra, 10+ minutes to install kodi.

Invictaz, 2017-05-06:
Is the MicroSD slot that Fast of the PiZero 1.2? 

Why do Packard need the absolute latest of revisions?

Fourdee, 2017-05-06:
@Invictaz 

> Is the MicroSD slot that Fast of the PiZero 1.2?

17.5MB/s R/W

But random read/write is highly improved on UHS-3 cards. You'll notice a vast reduction to install times/updates.

> Why do Packages need the absolute latest of revisions?

If we dont do this, and for example you install Kodi, which lets say requires gnupg2 2.0.26-6+deb8u1, it would fail to install without the upgrade beforehand.

k-plan, 2017-05-07:
> Is the MicroSD slot that Fast of the PiZero 1.2?

 RPi Zero W (armv6l) with DietPi on a [SanDisk Extreme PLUS 16 GB microSDHC](https://www.amazon.de/SanDisk-Extreme-microSDHC-Memory-Adapter/dp/B010NE3ND8)

[image]

With a little bit of [tweaking](http://dietpi.com/phpbb/viewtopic.php?f=15&t=331#p2605) (on one's own risk): 

[image]

Invictaz, 2017-05-11:
Interesting. But the W version might use a slightly faster card slot. Or memory management. Is that possible?

Invictaz, 2017-05-16:
@Fourdee @k-plan I tested it in v150. Not a super slow SD?

[image]

k-plan, 2017-05-17:
@Invictaz 

> Not a super slow SD?

Yes, for sure!

But I have one, it's more crappy:  [SanDisk microSDHC Ultra 4GB](https://www.amazon.de/SanDisk-microSDHC-Ultra-original-Handelsverpackung/dp/B001972AV0)

[image]

```
root@RPi-Zero:~# dmesg | grep mmc

[    1.841822] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[    2.050331] mmc0: host does not support reading read-only switch, assuming write-enable
[    2.064841] mmc0: new high speed SDHC card at address 86fc
[    2.075589] mmcblk0: mmc0:86fc SU04G 3.69 GiB
```

> Interesting. But the W version might use a slightly faster card slot. Or memory management. Is that possible?

No!

Device on all screenshot  is a RPi Zero (v1.2)

```
root@RPi-Zero:~# /DietPi/dietpi/misc/rpi_boardinfo

  - RPi Board Info -
 Revision         |     900093
 Released         |     Q2 2016
 Model            |     Zero
 Memory           |     512 MB
 Manufacturer     |     Sony
```

But with excact the same device, but wit a good sd-card it looks : 
[image]

> @Fourdee @k-plan I tested it in v150. 

All tests with the same DietPi Version : 

```
 ───────────────────────────────────────
 DietPi     | 20:32 | Wed 17/05/17
 ───────────────────────────────────────
 V150       | RPi Zero (armv6l)
 ───────────────────────────────────────
```
Difference make the sd-card:  [SanDisk Extreme PLUS 32 GB microSDHC U3 ](https://www.amazon.de/SanDisk-Extreme-microSDHC-Memory-Adapter/dp/B010NE3QS0?th=1
)

```
root@PPi-Zero_with-U3:~# dmesg | grep mmc

[    1.841814] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[    2.082102] mmc0: host does not support reading read-only switch, assuming write-enable
[    2.099103] mmc0: new high speed SDHC card at address aaaa
[    2.120235] mmcblk0: mmc0:aaaa SE32G 29.7 GiB
```

---------------------------------

And with a little bit of tweaking, it's look like this:
[image]

```
root@PPi-Zero_with-U3:~# dmesg | grep mmc

[    1.823705] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[    2.063980] mmc0: host does not support reading read-only switch, assuming write-enable
[    2.080620] mmc0: overclocking to 100000000Hz
[    2.099538] mmc0: new high speed SDHC card at address aaaa
[    2.110454] mmcblk0: mmc0:aaaa SE32G 29.7 GiB

```

Try to read [here](http://dietpi.com/phpbb/viewtopic.php?f=15&t=331#p2583) and [here](https://github.com/RetroPie/RetroPie-Setup/wiki/Overclocking).

But please don't try this with a cheap  non U3 sd-card !!!

Pro-Tip:
You speed problem is not DietPi, not the Version number, not the RPi Zero device.

Dump you card to waste bin and buy a really tested, good sd-card.

But you can do what ever you like ....

MichaIng, 2018-01-31:
I will close that issue. I think question was answered and for overall software quality it makes sense to regularly update the repo packages. On stable branches this is already relatively rarely and mostly just minor updates for stability, security,  performance, well tested with all other repo software packages.

closed 2018-01-31 by MichaIng