MichaIng/DietPi #930: Pizero W - wlan0 not found, eth0 not found

9H5G, 2017-05-06:
I've discovered an issue with PiZero W.  Was working fine for a month or so and took updates.  However, I decided to flash a new card and configure it for headless start as usual by enabling and setting the wlan parameters in dietpi.txt.  However, in the startup log, with this freshly flashed card, I see that Dietpi brings up neither network interface since it cannot find them (the interface).

Using a freshly flashed Raspian works just fine! #frustrating :)

What would you like me to do to track this one down?

Fourdee, 2017-05-06:
@9H5G 

>  I decided to flash a new card and configure it for headless start as usual by enabling and setting the wlan parameters in dietpi.txt. However, in the startup log, with this freshly flashed card, I see that Dietpi brings up neither network interface since it cannot find them (the interface).

Which image version of DietPi are you currently using? v148 is latest.

Fourdee, 2017-05-06:
@9H5G 

I just successfully completed a automated install (using WiFi creds set in dietpi.txt) on RPi Zero, with v148 image:
http://dietpi.com/downloads/images/DietPi_RPi-armv6-(Jessie).7z

Any chance you could download the above image and test it again?

9H5G, 2017-05-06:
r work.  Dan,

It seems that 148 worked so no dramas.  I was still on 143 since I was
travelling.

Thanks for all your work - drop me a note if you're ever over in Essex and
I'll buy you a beer :)

Kind Regards
John

Fourdee, 2017-05-06:
@9H5G 

Excellent. RPi Zero W support was added to v145 image and higher.

> Thanks for all your work - drop me a note if you're ever over in Essex and
I'll buy you a beer :)

Will do, thanks 👍

closed 2017-05-06 by Fourdee