MichaIng/DietPi #931: Diet Pi Image for Beagle Bone Black

lucario1001, 2017-05-07:
wondering if it can be made 
AM335x 1GHz ARM® Cortex-A8

also love your software

works well on the raspberry pi 👍

Fourdee, 2017-05-07:
@lucario1001 

Thanks for the request. DietPi would run really well on this single core board.

I did email beaglebone over 3 months ago, requesting dev sample boards, then we can implement a DietPi image. However, still no response as of yet.

Fourdee, 2017-05-07:
Lets give twitter a bash:
https://twitter.com/DietPi_/status/861230917322502144

lucario1001, 2017-05-07:
cool thanks 👍 soon hope we might have a image for it

pdp7, 2017-05-08:
@jadonk @robertcnelson what do you think?

scottwilliamsinaus, 2017-05-09:
+1 for this! BBB has a following amongst audiophiles for its ability to have the I2S output synced to an external master clock. Also, the Botic overlay (https://github.com/miero/botic-dev) on BBB is the only way to get hires multichannel output from a SBC at present (I think - happy to be corrected), so incorporating that would be fantastic. 
Brilliant work with DietPi otherwise! It has turned Linux from being a drag to fun, and saved me countless hours. Donation made!

iSsO, 2017-10-18:
+1
Anything we can do to accelerate this?

Fourdee, 2017-10-19:
@iSsO 
Ordered one on RS, should be here next week (back order)
[image]

iSsO, 2017-10-19:
Great news, I've just ordered the newly announced PocketBeagle one, it will arrive next week as well.

I think they are using the same processor, but the pocket is using a integrated module (Octavo) instead of the bare IC.

Do you think DietPi can work on the pocket as well?

I can help in testing if you need to.

lucario1001, 2017-10-21:
still have my beagle bone black if you need help testing

Fourdee, 2017-11-18:
Notes:
- Copy EMMC to SD > https://elinux.org/BeagleBone_Black_Extracting_eMMC_contents
- Copy SD to EMMC > http://derekmolloy.ie/write-a-new-image-to-the-beaglebone-black/

Fourdee, 2017-11-22:
### Performance:

- CPU
- Network
- Onboard EMMC:
[image]

Fourdee, 2017-11-22:
How to flash EMMC:
- Boot system with DietPi on SD card (hold S2 during power on for 2-5 seconds)
- Edit ```/DietPi/uEnv.txt``` and enable line at bottom (remove ```#```) ```cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh```
- Reboot system
- Wait for device to flash (LED's in up and down motion). Device will power down once its completed.
- Eject SD and power on.

Fourdee, 2017-11-22:
> https://twitter.com/DietPi_/status/933419127276036096

Completed. Image now live.

closed 2017-11-22 by Fourdee

antonellocaroli, 2018-09-20:
I can not find the image on the download page

Fourdee, 2018-09-20:
@antonellocaroli 

Apologies, the image is no longer available/maintained due to low user count. You can install DietPi on the system using PREP:
https://github.com/Fourdee/DietPi/issues/1285#issue-280771944

zhanglongqi, 2020-06-27:
@Fourdee Just curious about how you know the user count? what is the data source?

MichaIng, 2020-06-27:
E.g. the survey statistics give/gave a hint: https://dietpi.com/survey/#device
See for "BBB", two reported systems, probably one was Fourdees testing system. Since there is no BBB image download available anymore this cannot be compared well now, but it was when Fourdee closed this ticket.

But you can always use https://github.com/MichaIng/DietPi/blob/dev/PREP_SYSTEM_FOR_DIETPI.sh to migrate from (mostly) any Debian-based image to DietPi.

LaurensVanAcker, 2021-08-13:
I tried this script one a BeagleBone Black rev C, but the connection drops at `Removing connman (1.36-2) ...`.

RobertCNelson, 2021-08-13:
@LaurensVanAcker that's expected, all the official BeagleBoard.org images use connman as the default network connection tool..

MichaIng, 2021-08-13:
Yes, with connman and NetMan this is a known issue: #2054
For some reason, on uninstall, those completely deconfigure/break the current network connection. So `ifupdown` needs to be installed, `/etc/network/interfaces` configured and then either after an `ifup --force eth0`/`ifup --force wlan0` it's already possibly to uninstall connman/NetMan, or it needs to be disabled and then a reboot done to have `ifupdown` taking over:
```
systemctl disable connman
systemctl enable networking
reboot
```

jnk19, 2022-02-04:
Any update on this? I use the build script everything finish normal ( I had pre removed connman and config `/etc/network/interfaces`) but after reboot I have "black screen" and no more connection with SSH. Anyone have resolve this?

MichaIng, 2022-02-04:
Which base image did you use? The Debian 1GB console images seem to be the best start: https://beagleboard.org/latest-images

Can you also show the full output of DietPi-PREP?

What I recognised is that there is a custom APT source in the main sources.list. You should add it before running PREP as dedicated source:
```sh
echo 'deb https://repos.rcn-ee.com/debian/ buster main' > /etc/apt/sources.list.d/bbb.conf
```
Replace `buster` with `bullseye` if you plan to upgrade. After PREP, before rebooting, you may want to reinstall the packages which are shipped by the image:
```
apt install bb-bbai-firmware bb-cape-overlays bb-customizations bb-wl18xx-firmware
```
It adds Bluetooth and WiFi firmware (from what I understand, though BBB doesn't have any of both, does it?) and a bunch of services and scripts where I'm not sure which one may be required (one does something on U-Boot configs on kernel upgrade).

You do not have a UART adapter to watch early boot logs from an external console, do you?

RobertCNelson, 2022-02-04:
I've moved these to our forum's.: https://forum.beagleboard.org/tag/bbbio_debian

The Bullseye images now use systemd-networkd vs connman..  (Debian no longer has a maintainer for connman..)

Regards,

jnk19, 2022-02-04:
I use this image: AM3358 Debian 10.3 2020-04-06 4GB eMMC IoT Flasher 

No BBB do not have Bluetooth or Wireless (except the Wireless version I do not use that) so probably not need these packages you mention on you post.

No unfortunately I do not have a UART adapter, but I can get the output of DietPI-PREP.

I will try the new images with bullseye release that @RobertCNelson provided and I will try to build there also.

I hope all goes well.

RobertCNelson, 2022-02-04:
That IoT image has a lot of extra stuff.. Node-RED, CLoud9, etc.. it would be best to start with a console bsae, you want to use DietPi's scripts..

MichaIng, 2022-02-04:
Many thanks for chiming in. Great that there are Bullseye snapshots already. Should then be preferred for a DietPi-PREP run. `systemd-networkd` does not abort the network connection when stopped, so that issue would then be solved for BBB.

Awesome to see that recent Linux versions are provided, this is very rare among SBC manufacturers.

Having a look through the latest am335x Bullseye snapshot:
- @RobertCNelson would you consider moving the `repos.rcn-ee.com` repository into a dedicated file in `/etc/apt/sources.list.d`? This is mostly a convention to have only the major distribution repository (Debian) in `sources.list` and others in dedicated files.
- There is not a U-Boot package `bb-u-boot-am335x-evm` which one may want to install as well. I guess it does not automatically flash U-Boot on upgrade, but provides the new images and a script to flash them.
- `bb-customizations`, mentioned above, is btw an important one, which updates the symlinks used by the bootloader to kernel, dtb and initramfs on kernel upgrades. Though, since there is no kernel meta package (at least not pre-installed), the kernel currently needs to be upgraded manually:
    ```sh
    apt search 'linux-image-.*-ti-r[0-9]'
    ```

giusCB, 2022-03-02:
hi guys, i've a little stock of BB Black. 
Can someone summarize how to flash dietpi system on emmc memory ?
(amny) Thanks in advance

Joulinar, 2022-03-02:
Does this blog might be helpful https://dietpi.com/blog/?p=1236

giusCB, 2022-03-02:
thanks for the reply,
there is something more specifically for the BBB ?

Joulinar, 2022-03-02:
nope