MichaIng/DietPi #932: Problem installing LESP webserver stack

Zac610, 2017-05-07:
Hi, I am having some troubles installing the LESP stack on a RPi2; 

I had an older version of DietPi installed (about 5 versions older); I launched `dietpi-software` for LESP installing but it failed; so I updated to latest dietpi version but still the same problem:
```
Apt-get has failed:
  - Apt string = nginx xml-core --no-install-recommends
  - Error code = 100
  - Software Title = Nginx: webserver
  - Apt log = /var/log/dietpi-software_apt.log
```
and here the Apt log:
```
Setting up nginx-full (1.6.2-5+deb8u4) ...
Job for nginx.service failed. See 'systemctl status nginx.service' and 'journalctl -xn' for details.
invoke-rc.d: initscript nginx, action "start" failed.
dpkg: error processing package nginx-full (--configure):
 subprocess installed post-installation script returned error exit status 1
dpkg: dependency problems prevent configuration of nginx:
 nginx depends on nginx-full (>= 1.6.2-5+deb8u4) | nginx-light (>= 1.6.2-5+deb8u4) | nginx-extras (>= 1.6.2-5+deb8u4); however:
  Package nginx-full is not configured yet.
  Package nginx-light is not installed.
  Package nginx-extras is not installed.
 nginx depends on nginx-full (<< 1.6.2-5+deb8u4.1~) | nginx-light (<< 1.6.2-5+deb8u4.1~) | nginx-extras (<< 1.6.2-5+deb8u4.1~); however:
  Package nginx-full is not configured yet.
  Package nginx-light is not installed.
  Package nginx-extras is not installed.

dpkg: error processing package nginx (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 nginx-full
 nginx

```

Fourdee, 2017-05-07:
@Zac610 

Thanks for the report 👍 

I was unable to replicate on a fresh installation with RPi, so must be limited to your existing installation and package combination, throwing this out.

Before we attempt a fix, lets check install software on system, please paste results:
```
dietpi-software list | grep ' =2 '
```

Zac610, 2017-05-07:
@Fourdee 

Thanks for working this out! 

Here's the output of software list:
```
id 3 | =2 | mc: midnight commander, powerful file manager |
id 5 | =2 | alsa: linux sound system |
id 16 | =2 | build-essentials: common packages for compile |
id 17 | =2 | git client: git clone etc |
id 39 | =2 | minidlna: media streaming server (dlna, upnp) | +alsa | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p49
id 42 | =2 | plex media server: web interface media streaming server | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1949#p1949
id 44 | =2 | transmission: bittorrent server with web interface (c) | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p46
id 96 | =2 | samba: feature-rich file server | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=10#p56
id 103 | =2 | dietpi-ramlog: minimal, optimized logging | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p68
id 104 | =2 | dropbear: ssh server | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=10#p62

```
and here is the bug report reference code: `733f7397-cbfd-409b-9e21-8f96399046e4`

Fourdee, 2017-05-08:
@Zac610 

Thanks,

Should be safe to remove nginx and try reinstall. Please, do the following:
```
apt-get purge nginx nginx-*

#Install LESP
dietpi-software install 78
```

If this still fails with same error, I'll need to see the following:
```
cat /etc/apt/sources.list
ls -lha /etc/apt/sources.list.d/*
cat /etc/apt/sources.list.d/*
```

Zac610, 2017-05-09:
Still same problem! 

I investigated further, it seems it is a problem of disabled IPv6 on my RPi2 conflicting with IPv6 port listening in default nginx configuration. Main source [here](https://askubuntu.com/a/801298/320709) and [here](https://etc.banana.fish/?p=75).

btw here are the requested outputs:
```
root@Muletto:~# cat /etc/apt/sources.list
deb http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian jessie main contrib non-free rpi

root@Muletto:~# ls -lha /etc/apt/sources.list.d/*
-rw-r--r-- 1 root root  52 Mar  6 15:39 /etc/apt/sources.list.d/plex.list
-rw-r--r-- 1 root root 193 Feb 25 16:12 /etc/apt/sources.list.d/raspi.list

root@Muletto:~# cat /etc/apt/sources.list.d/*
deb [arch=armhf] http://dev2day.de/pms/ jessie main
deb http://archive.raspberrypi.org/debian/ jessie main ui
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ jessie main ui
```

Fourdee, 2017-05-09:
@Zac610 

> I investigated further, it seems it is a problem of disabled IPv6 on my RPi2 conflicting with IPv6 port listening in default nginx configuration. Main source here and here.

Good find 👍 

Solution, enable IPv6, install LESP, then disable IPv6 on system and config if required?

Fourdee, 2017-05-27:
Marking as known issue, please reopen if required.

closed 2017-05-27 by Fourdee