MichaIng/DietPi #934: Installed PiVPN is not working

enolp, 2017-05-07:
After installing PiVPN I realized that all .ovpn files are not generated correctly and when I create them, they cannot be copied into the user's home folder because the path is wrong.

`root@Server:~# pivpn -a
cat: /etc/pivpn/INSTALL_USER: No such file or directory
Enter a Name for the Client:  test
Enter the password for the client:  
Enter the password again to verify:  
spawn ./easyrsa build-client-full test

Note: using Easy-RSA configuration from: ./vars
Generating a 2048 bit RSA private key
.........................................................................................+++
.....................+++
writing new private key to '/etc/openvpn/easy-rsa/pki/private/test.key.UocOuZcqkN'
Enter PEM pass phrase:
Verifying - Enter PEM pass phrase:
-----
Using configuration from /etc/openvpn/easy-rsa/openssl-1.0.cnf
Check that the request matches the signature
Signature ok
The Subject's Distinguished Name is as follows
commonName            :ASN.1 12:'test'
Certificate is to be certified until May  5 15:08:54 2027 GMT (3650 days)

Write out database with 1 new entries
Data Base Updated
Client's cert found: test.crt
Client's Private Key found: test.key
CA public Key found: ca.crt
tls-auth Private Key found: ta.key
cat: Default.txt: No such file or directory
cp: cannot create regular file ‘/home//ovpns/test.ovpn’: No such file or directory
chown: cannot access ‘/home//ovpns/test.ovpn’: No such file or directory

========================================================
Done! test.ovpn successfully created! 
test.ovpn was copied to:
  /home//ovpns
for easy transfer.
========================================================
`

Fourdee, 2017-05-07:
@enolp 

Thanks for report 👍 

> /home//ovpns/test.ovpn

Looks like a possible bug in the PiVPN install script, ```//```. But i'll run a test install here to verify.

Fourdee, 2017-05-07:
@enolp 

Strange, just did a test install, client profiles created fine:
```
Using configuration from /etc/openvpn/easy-rsa/openssl-1.0.cnf
Check that the request matches the signature
Signature ok
The Subject's Distinguished Name is as follows
commonName            :ASN.1 12:'test'
Certificate is to be certified until May  5 17:34:09 2027 GMT (3650 days)

Write out database with 1 new entries
Data Base Updated
Client's cert found: test.crt
Client's Private Key found: test.key
CA public Key found: ca.crt
tls-auth Private Key found: ta.key

========================================================
Done! test.ovpn successfully created!
test.ovpn was copied to:
  /home/pivpn/ovpns
for easy transfer.
```

Try pulling the created client profile from:
https://github.com/pivpn/pivpn/blob/master/scripts/makeOVPN.sh#L173

```
cp /etc/openvpn/easy-rsa/pki/private/test.ovpn "$HOME"/test.ovpn
```

enolp, 2017-05-07:
@Fourdee 

´´´cp /etc/openvpn/easy-rsa/pki/private/test.ovpn "$HOME"/test.ovpn´´´ It didn't work.

Have you checked the generated .ovpn file? Because mine has only the certificate and key chains inside, all the other connection details are missing.

Yesterday I updated my Raspberry Pi firmware to the latest version, I hope it is not the failure.

closed 2017-05-24 by Fourdee