MichaIng/DietPi #936: Raspberry Pi 4.9 Kernel

Invictaz, 2017-05-08:
Has official Pi Zero support and Orange Pi and XFS improvements etc etc

http://www.phoronix.com/scan.php?page=news_item&px=Linux-4.9-ARM-Pull

Invictaz, 2017-05-11:
@Fourdee they do have 4.11

Fourdee, 2017-05-11:
@Invictaz 

Yep, however, 4.9 is LTS: https://www.phoronix.com/scan.php?page=news_item&px=Linux-4.9-LTS-Kernel

Fourdee, 2017-05-12:
Notes:
4.9 kernel contains required Allo DAC's already. Only change we needed to make is +384KHz support in ALSA.

Invictaz, 2017-05-13:
@fourdee So what to do if you want to run the 3D apps supported in 4.11 then? I don't get the point of long term support if updates are available and they don't break anything as most likely backwards compatible.

Fourdee, 2017-05-14:
@Invictaz 

> So what to do if you want to run the 3D apps supported in 4.11 then?

You can always compile 4.11 and install it on your DietPi system for testing purposes.

Fourdee, 2017-05-14:
[image]

Fourdee, 2017-05-14:
OpenGL should now be functional, lets test:
https://github.com/Fourdee/DietPi/issues/834

Fourdee, 2017-05-15:
Completed. 4.9 kernel patched in v150 update.

closed 2017-05-15 by Fourdee