MichaIng/DietPi #937: WiringPi for BananaPi/BananaPi Pro missing

Elluminatus, 2017-05-09:
Hi, i tried to install wiringPi via the Software Setup of DietPi on my BananaPi/ BananaPi Pro. But i can´t see the entry. I installed it manually on an other bananaPiBoard and it`s working fine. So can you implement it on the BananaPi Pro Software Setup?

Greeting Elluminatus

Fourdee, 2017-05-10:
@Elluminatus 

Thanks for request 👍 

I'll take a look when possible. note: https://github.com/LeMaker/WiringBP

Fourdee, 2017-05-10:
Completed, available in v150:
```
root@DietPi:~# gpio readall
 +-----+-----+---------+------+---+--Banana Pro--+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT5 | 1 |  3 || 4  |   |      | 5V      |     |     |
 |   3 |   9 |   SCL.1 | ALT5 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 0 |  7 || 8  | 0 | ALT0 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 0 | ALT0 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 | ALT4 | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 | ALT4 | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 | ALT4 | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI | ALT5 | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO | ALT5 | 0 | 21 || 22 | 0 | ALT4 | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK | ALT5 | 0 | 23 || 24 | 0 | ALT5 | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 0 | ALT5 | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 | ALT4 | 0 | 27 || 28 | 0 | ALT4 | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 0 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 | ALT4 | 0 | 31 || 32 | 0 | ALT4 | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 | ALT5 | 0 | 35 || 36 | 0 | ALT5 | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 | ALT5 | 0 | 37 || 38 | 0 | ALT5 | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | ALT5 | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+--Banana Pro--+---+------+---------+-----+-----+
```

closed 2017-05-10 by Fourdee