MichaIng/DietPi #938: BananaPi pro image running on BananaPi too

Elluminatus, 2017-05-09:
The BananaPi Pro image of DietPi is running on BananaPi Pro image fine. So maybe you can write this in your download section as information for other users.

Greetings Elluminatus

Invictaz, 2017-05-10:
They are both Allwinner A20 so quite compatible :)

Fourdee, 2017-05-10:
> The BananaPi Pro image of DietPi is running on BananaPi Pro image fine. So maybe you can write this in your download section as information for other users.

Thanks for info 👍 

I'll link this info to the BPi image request ticket for now: https://github.com/Fourdee/DietPi/issues/824

Elluminatus, 2017-05-10:
Maybe you can add a picture from the BananaPi to the download section or a hint under the bananaPi pro picture?

Fourdee, 2017-05-10:
@Elluminatus 

> Maybe you can add a picture from the BananaPi to the download section or a hint under the bananaPi pro picture?

Although same CPU, they some differences with hardware and features. The kernel/modules and voltages can be unique for each image/board. Potential damage by using an incorrectly configured image for the board (if CPU/Dram voltages are different for each board).

I'll leave the information on Git for now.

Elluminatus, 2017-05-10:
@ Fourdee

Thank you for the hint, I will make diethers tests.

closed 2017-05-10 by Elluminatus