MichaIng/DietPi #940: DB drivers problem on Nextcloud fresh install

enolp, 2017-05-09:
Installing Nextcloud on a Raspberry Pi 2 shows me an error when you try to setup your installation within the webapp. It says: 

```No database drivers (sqlite, mysql, or postgresql) installed.```

Right now I'm using lighttpd as a web server.

Fourdee, 2017-05-09:
@enolp 

Looks like a possible issue with failed mysql installation.

Can you please send us a bug report? I'd like to check system settings/packages:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=499#p2184

enolp, 2017-05-09:
Bug report done and sent. Here is the reference code: 2d1f034b-8d3b-4009-9a78-3c5858f120e8-0 .

Thank you for your support.

Fourdee, 2017-05-10:
- fail2ban
- llmp
- pihole
- nextcloud
- pivpn

> No database drivers (sqlite, mysql, or postgresql) installed.

MySQL and all php modules appeared installed correctly. Possibly a failed installation.

#### Ok, lets try purging nextcloud and mysql, then reinstall nextcloud:

_If system is critical, back it up 1st:_
```
dietpi-backup
```

Then:
```
#You'll receive some errors regarding mysql not found on nextcloud removal, thats fine as DB is removed during mysql removal.
dietpi-software uninstall 86 114 

#Install Nextcloud
dietpi-software install 114
```

enolp, 2017-05-10:
I did that reinstallation but it didn't work, however I uninstalled mysql and switched to mariadb, and Nextcloud seems to be working. The problem now is I don't know which username and password were set.

Fourdee, 2017-05-11:
@enolp 

> The problem now is I don't know which username and password were set.

Please see online doc:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=3026#p3026

Fourdee, 2017-05-15:
@enolp 

Is your nextcloud installation now functional/resolved?

enolp, 2017-05-27:
Sorry for not replying but yes, all worked fine. I did a fresh install and used Lighttpd, MariaDB and PHP as my choice.

Fourdee, 2017-05-28:
@enolp 

Thanks. I'll mark this as resolved. Please reopen if required.

closed 2017-05-28 by Fourdee