MichaIng/DietPi #941: No Cpu temperature after 141 to 149 update

stdusm, 2017-05-09:
After latest update from v141 to v149 CPU temperature is not working. 
Show always 0 C'.

>   ─────────────────────────────────────────────────────
>  DietPi CPU Info
>  Use dietpi-config to change CPU / performance options
>  ─────────────────────────────────────────────────────
>  Architecture |     armv7l
>   **Temp         |     0'c** | Who put me in the freezer!
>  Governor     |     powersave
> 
>   Current Freq    Min Freq   Max Freq
>  CPU0         |      480 MHz      480 MHz    912 MHz
>  CPU1         |      480 MHz      480 MHz    912 MHz
>  CPU2         |      480 MHz      480 MHz    912 MHz
>  CPU3         |      480 MHz      480 MHz    912 MHz
>  CPU4         |      480 MHz      480 MHz    912 MHz
>  CPU5         |      480 MHz      480 MHz    912 MHz
>  CPU6         |      480 MHz      480 MHz    912 MHz
>  CPU7         |      480 MHz      480 MHz    912 MHz

Fourdee, 2017-05-09:
Hi,

Thanks for report 👍 

Known issue, fixed in v150. You can apply a manual fix now with link below:
https://github.com/Fourdee/DietPi/issues/909#issuecomment-298228895

stdusm, 2017-05-09:
Thanks a lot !

closed 2017-05-09 by stdusm