MichaIng/DietPi #942: [Out off topic] Does the DietPi forum work?

it-praktyk, 2017-05-09:
Hi

I tried register to the DietPi forum twice - at: May 2th and today - but I didn't receive registration acceptation.

I would like ask for advise about what is the better/easiest option to prepare the new one package - Ansible managed node prerequisites.

Fourdee, 2017-05-10:
@it-praktyk 

> I tried register to the DietPi forum twice - at: May 2th and today - but I didn't receive registration acceptation.

Apologies for that. Must of been triggered as a possible spambot account.

Whats the username for the account you made today? I'll activate.

> I would like ask for advise about what is the better/easiest option to prepare the new one package - Ansible managed node prerequisites.

Sorry, not sure what you mean? Or is this request and you'd like to see "Ansible" in ```dietpi-software```?

it-praktyk, 2017-05-10:
>> I  would like ask for advise about what is the better/easiest option to prepare the new one package - Ansible managed node prerequisites.

> Sorry, not sure what you mean? Or is this request and you'd like to see "Ansible" in dietpi-software?

I can try add it myself - probably Python 2.7.x + some modules from pip will be needed + tests with dropbear - but I need some instruction when I should start .

Packages are define in
```
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software
```
?

Thx for remarks about reasons why my email was classified as spam.

Fourdee, 2017-05-10:
@it-praktyk 

Python-pip install:
```
dietpi-software install 130
```

Then try installing Ansible via apt: http://docs.ansible.com/ansible/intro_installation.html#id18

Elluminatus, 2017-05-10:
Same issue here: still waiting for activation: Name: Elluminatus

Fourdee, 2017-05-11:
@Elluminatus 

Your forum account is now active.

Shmio, 2017-05-18:
Same issue.  Name: Shmio

Appreciate it.

Fourdee, 2017-05-19:
@Shmio 

All pending forum accounts now enabled.

Fourdee, 2017-06-08:
I'll mark this as closed.

I aim to check (and enable) the forum accounts for activation every day/2 days, where possible.

As for possible spam accounts, avoid using names such as ```exmdkfwiemfua``` and email addresses such as ```gooooieea..xoeckx@lol.unknownwebsite.z```

closed 2017-06-08 by Fourdee