MichaIng/DietPi #946: Broken Subsonic 6 installation

Magtula, 2017-05-10:
It's not possible to install Subsonic 6. I tried it also with a new installed dietpi-system.
I attached an error-screenshot and log-File.

[Log.txt](https://github.com/Fourdee/DietPi/files/991030/Log.txt)
[image]

Fourdee, 2017-05-11:
@Magtula 

Thanks for the report 👍 

I'll take a look, which board?

Fourdee, 2017-05-11:
Package conflicting:
```
dpkg: error processing archive /var/cache/apt/archives/lame_3.99.5+repack1-7+deb8u1_armhf.deb (--unpack):
 trying to overwrite '/usr/bin/lame', which is also in package libmp3lame 3.99-1
```

Fourdee, 2017-05-11:
Fixed for v150.

@Magtula Manual fix for this now:
```
apt-get install -f
apt-get purge lame -y
sed -i '/AGI lame/d' /DietPi/dietpi/dietpi-software

#Install subsonic 6
dietpi-software install 34
```

Magtula, 2017-05-12:
Thank you very much - Awesome support!

Just for the sake of completeness, the board is a Raspberry Pi 2.

Fourdee, 2017-05-12:
@Magtula 
I'll mark this as resolved, please reopen if required.

closed 2017-05-12 by Fourdee