MichaIng/DietPi #947: Cannot connect by sftp to Dietpi

TitaneBoy, 2017-05-11:
Hi Team..

I was trying to connect by sftp (with WinScp or FileZilla) to my Dietpi installed in my raspberry Pi 3 Model B. Unfortunetly, I can't connect in any way. WinSCP tells me that STFP Packet is too big and FileZilla is not able to connect to my pi.

Can you help me please ?

Thank you

Fourdee, 2017-05-11:
@TitaneBoy 

Hi,

Which software installed on RPi, Proftpd? And, does standard FTP function?

3d-gussner, 2017-05-11:
You have to install OpenSSH instead of default Dropbear
Open 'dietpi-software' and go to SSH server and there choose OpenSSH

TitaneBoy, 2017-05-11:
@Fourdee :  Hi !

My installed sofware are: 
-OpenSSH (client and server)
-Node.js
-iftop
-IPTraf
-Git Client
-Gogs
-LLMP: Lighttpd | mysql |php
-Lighttpd: webserver
-MySQL: database
-PHP : webserver
-OpenVPN: vpn server
-Rsyslog: system logging
-DietPi-Ramlog
-NextCloud
-Webmin
-PiVPN
-Python Pip

I'm able to connect to my dietpi with the windows software "puTTY" (so by SSH) but not by SFTP with WinSCP or FileZilla

@3d-gussner : As I said before, OpenSSH is already installed and I'm able to connect to DietPi with puTTY.

Fourdee, 2017-05-11:
@TitaneBoy 

Try the SCP protocol:
[image]

http://www.jscape.com/blog/scp-vs-sftp

Fourdee, 2017-05-11:
@TitaneBoy 

Strange, SFTP also works for me straight out the box.

Please check this package is installed:
```
dpkg -l | grep openssh-sftp-server
```

Also, try disabling jumbo ethernet packets, if enabled on your Windows PC.

TitaneBoy, 2017-05-11:
@Fourdee By running `dpkg -l | grep openssh-sftp-serverThe`, I can see that package openssh-sftp-server is  already installed...This is the output of the command:

`ii  openssh-sftp-server        1:6.7p1-5+deb8u3           armhf        secure shell (SSH) sftp server module, for SFTP access from remote machines`

Also, the jumbo frame is already disabled too. I never touched that settings...

This is what I can see in /var/log/auth.log just after I'm trying to connect with WinSCP on my pi, by sftp, for user dietpi
```
May 11 11:20:13 MyDietPi sshd[28003]: Accepted password for dietpi from XXX.XXX.XXX.XX port 42297 ssh2
May 11 11:20:13 MyDietPi sshd[28003]: pam_unix(sshd:session): session opened for user dietpi by (uid=0)
May 11 11:20:13 MyDietPi sudo:   dietpi : TTY=unknown ; PWD=/home/dietpi ; USER=root ; COMMAND=/DietPi/dietpi/login
May 11 11:20:13 MyDietPi sudo: pam_unix(sudo:session): session opened for user root by (uid=0)
May 11 11:20:13 MyDietPi sudo: pam_unix(sudo:session): session closed for user root
May 11 11:20:13 MyDietPi sshd[28003]: pam_unix(sshd:session): session closed for user dietpi
```
By trying SCP protocol, it looks to work and I can remote my dietpi.

What I have to do in order to make it work with SFTP protocol ?

Thank you for your answers

Fourdee, 2017-05-12:
@TitaneBoy 

> https://winscp.net/eng/docs/message_large_packet

The above mentions text print on server, during login can be a cause, so lets check these:
```
cat "$HOME"/.bashrc
cat "$HOME"/.profile
```

TitaneBoy, 2017-05-16:
Hi @Fourdee .

By disabling `#/DietPi/dietpi/login` from `"$HOME"/.bashrc`, it looks that I can finally connect with sftp protocol with WinSCP  :-)

By reading the link you provided, does it mean I have to move the `/DietPi/dietpi/login` from `"$HOME"/.bashrc` to `"$HOME"/.profile` ?

Thanks again for your answer...

Fourdee, 2017-05-17:
@TitaneBoy 

> By disabling #/DietPi/dietpi/login from "$HOME"/.bashrc, it looks that I can finally connect with sftp protocol with WinSCP :-)

Excellent, great to hear 👍 

> By reading the link you provided, does it mean I have to move the /DietPi/dietpi/login command from "$HOME"/.bashrc to "$HOME"/.profile ?

For now, yes. 
However, if you can please confirm this is functional (with SFTP and login script executes) once moved, we'll roll this change to DietPi.

Fourdee, 2017-06-17:
Marking as resolved. Please reopen if required.

closed 2017-06-17 by Fourdee

TitaneBoy, 2017-06-17:
Hi.. Sorry for the delay... I confirm that your given solution worked for me.. 😊

Thank you

vvv-ibm, 2018-03-20:
It seems that the login command is still in bashrc and it still causes issues with scp? Is there going to be a fix for this?
Thanks!

Fourdee, 2018-03-20:
@vvv-ibm 

Hi,

Which SSH server installed and client used?

vvv-ibm, 2018-03-20:
openssh-server              1:7.4p1-10+deb9u3
openssh-client                1:7.6p1-4 
dietpi 6.4

KaKi87, 2024-04-17:
Hi,

Sorry for necroposting, but this additional information might help people finding this answer.

> You have to install OpenSSH instead of default Dropbear Open 'dietpi-software' and go to SSH server and there choose OpenSSH

This answer solves the following issues :

- "Received unexpected end-of-file from SFTP server" on FileZilla ;
- Freeze after login on Midnight Commander ;
- Generic error on GNOME Files.

Also, note that it changes the SSH signature so it must be removed if it was accepted : `ssh-keygen -f "/home/<username>/.ssh/known_hosts" -R "<server_hostname>"`

Thanks

MichaIng, 2024-04-18:
Indeed, with Dropbear as SSH server I also face connection drops on larger transfer at about every 450 MiB: https://github.com/MichaIng/DietPi/issues/3296#issuecomment-1382730526
With WinSCP you can however continue the transfer where it stuck before.

Would you mind to open a new issue about MC and GNOME Files SFTP connection issues? I would like to try replicating it when I find time. I am not aware of any SFTP connection issues with Dropbear and any SFTP client so far.

KaKi87, 2024-04-18:
I wasn't reporting a new problem actually, just making the reason and consequences more visible, basically saying : in order to have SFTP on DietPi, just replace Dropbear with OpenSSH.

> I am not aware of any SFTP connection issues with Dropbear and any SFTP client so far.

What ? Dropbear doesn't support SFTP, that's known, even yourself mentioned it on other issues.

That said, making OpenSSH default instead of Dropbear would avoid users experiencing this altogether.

KaKi87, 2024-04-18:
[image]

[image]

> Dropbear supports SFTP perfectly fine (when installing any SFTP server, like the dedicated OpenSSH SFTP server package or GreenEnd SFTP server)

Well, whatever it is, SFTP working by default on DietPi would be nice.

Thanks

MichaIng, 2024-04-18:
Ah, I misinterpreted context in this thread. Of course you require an SFTP server executable (OpenSSH does as well):
```sh
apt install openssh-sftp-server
```
Or, the one I would prefer:
```sh
apt install gesftpserver
ln -s /usr/libexec/gesftpserver /usr/lib/sftp-server # the default path most SFTP clients look for
```
I guess this would solve your issues while using Dropbear then. Of course switching to OpenSSH is a legit alternative.

While I would not change the default SSH server, some visual option to install an SFTP server, when Dropbear is selected, would be helpful, indeed.

KaKi87, 2024-04-18:
> While I would not change the default SSH server, some visual option to install an SFTP server, when Dropbear is selected, would be helpful, indeed.

And enabled by default, please.