MichaIng/DietPi #949: Image request for Odroid U3

Elluminatus, 2017-05-11:
Is it possible to get an image for the odroid u3?

Invictaz, 2017-05-11:
Samsung Exynos based board. Interesting, as on XDA developers there isn't much open source about these boards. A lot needs to be reverse engineered, especially GPU drivers. Maybe for the Odroid it's better.

Specs:

http://odroid.com/dokuwiki/doku.php?id=en:u3_hardware

Fourdee, 2017-05-12:
Very unlikely as this product is now discontinued. 
If I can get my hands on a U3, i'll do the image, if not we'd need someone with the board to prep a Debian image for DietPi:
https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh

pcmvanveen, 2017-10-09:
I own a U3. This is currently under production for an update to 16.04. So I'll try it out.

sovamakarosh, 2017-10-31:
Hello. How is the work going?

pcmvanveen, 2017-11-01:
It failed, I'm running odroid 16.04

volgende geschreven:

Fourdee, 2018-04-05:
Marking as closed due to:
- Product discontinued
- PREP offers a method to convert pre-image of debian into DietPi: https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh

closed 2018-04-05 by Fourdee