MichaIng/DietPi #950: Cannot set shairport-sync name attribute

vd, 2017-05-11:
When i update /etc/shairport-sync.conf with the name attribute it is not reflected when restarting all services with `dietpit-services restart`

Fourdee, 2017-05-12:
@vd 
Confirmed:
```
name = "Front Room";
```

Has no effect.

I believe its because we built Shairport-sync with avahi. When changing the hostname of the DietPi system, this also effects the shairport-sync name.

Please try changing the hostname:
```dietpi-config``` > Security Options > Change Hostname.

Fourdee, 2017-05-24:
@vd 

```name=``` having no effect will be fixed in v151:
https://github.com/Fourdee/DietPi/issues/962#issuecomment-301735237

```/usr/local/etc/shairport-sync.conf``` is the correct location for shairport-sync config.

I'll mark this as closed.

closed 2017-05-24 by Fourdee