MichaIng/DietPi #951: NTPD: Waiting for completion of time sync (20/20)

albaldi, 2017-05-13:
Hi, just installed dietpi
When I log in I get:
 [Info] Initializing database, please wait...
...
Welcome to DietPi-Software 
[Info] NTPD: Waiting for completion of time sync (1/20)
...
[Info] NTPD: Waiting for completion of time sync (20/20)
```
┌─────────────────────────────┤ Error ├──────────────────────────────┐
│                                                                    │
│ NTPD: System time has not been synced/updated correctly.           │
│                                                                    │
│ To prevent further issues with software installations,             │
│ DietPi-Software will now exit.                                     │
│                                                                    │
│ If problems persist, check the NTPD log:                           │
│    /var/log/ntpd.log                                               │
│                                                                    │
│ To override this error, run the following command:                 │
│    echo 0 > /etc/dietpi/.ntpd_exit_status                          │
│                                                                    │
│                                                                    │
│                                                                    │
│                               <Ok>                                 │
│                                                                    │
└────────────────────────────────────────────────────────────────────┘

```
log
```
 cat /var/log/ntpd.log
 6 Feb 15:22:35 ntpd[526]: proto: precision = 1.000 usec
 6 Feb 15:22:35 ntpd[526]: Listen and drop on 0 v4wildcard 0.0.0.0 UDP 123
 6 Feb 15:22:35 ntpd[526]: Listen and drop on 1 v6wildcard :: UDP 123
 6 Feb 15:22:35 ntpd[526]: Listen normally on 2 lo 127.0.0.1 UDP 123
 6 Feb 15:22:35 ntpd[526]: Listen normally on 3 eth0 192.168.1.11 UDP 123
 6 Feb 15:22:35 ntpd[526]: Listen normally on 4 lo ::1 UDP 123
 6 Feb 15:22:35 ntpd[526]: Listen normally on 5 eth0 fe80::ba27:ebff:fe5a:f195 UDP 123
 6 Feb 15:22:35 ntpd[526]: peers refreshed
 6 Feb 15:22:35 ntpd[526]: Listening on routing socket on fd #22 for interface updates
 6 Feb 15:40:31 ntpd[526]: Deleting interface #5 eth0, fe80::ba27:ebff:fe5a:f195#123, interface stats: received=0, sent=0, dropped=0, active_time=1070 secs
 6 Feb 15:40:31 ntpd[526]: Deleting interface #3 eth0, 192.168.1.11#123, interface stats: received=0, sent=48, dropped=0, active_time=1070 secs
 6 Feb 15:40:31 ntpd[526]: 185.19.184.35 interface 192.168.1.11 -> (none)
 6 Feb 15:40:31 ntpd[526]: 212.45.144.206 interface 192.168.1.11 -> (none)
 6 Feb 15:40:31 ntpd[526]: 213.251.52.250 interface 192.168.1.11 -> (none)
 6 Feb 15:40:31 ntpd[526]: 94.177.172.69 interface 192.168.1.11 -> (none)
 6 Feb 15:40:31 ntpd[526]: peers refreshed
 6 Feb 15:40:37 ntpd[526]: Listen normally on 6 eth0 192.168.1.11 UDP 123
 6 Feb 15:40:37 ntpd[526]: Listen normally on 7 eth0 fe80::ba27:ebff:fe5a:f195 UDP 123
 6 Feb 15:40:37 ntpd[526]: peers refreshed

```
current time on my machine
```
# timedatectl 
      Local time: Mon 2017-02-06 17:05:28 UTC
  Universal time: Mon 2017-02-06 17:05:28 UTC
        RTC time: n/a
       Time zone: Etc/UTC (UTC, +0000)
     NTP enabled: no
NTP synchronized: yes
 RTC in local TZ: no
      DST active: n/a
```
yep, I can travel back in time. :smile: 

What should I do?
Can I manually set time with:
`timedatectl set-time "yyyy-MM-dd hh:mm:ss"`
?

thanks

p.s.
sorry if is a forum's question, I'm still waiting to be approved by an administrator

Fourdee, 2017-05-14:
@albaldi 

Thanks for the report 👍 

> Can I manually set time with:
timedatectl set-time "yyyy-MM-dd hh:mm:ss"
?

If NTPD isn't working for your system, try changing the time system to SystemD in:
```dietpi-config``` > Advanced > Time sync mode >  Daemon + Drift

Check status with:
```
timedatectl status
```

```dietpi-software``` will then skip the check.

> sorry if is a forum's question, I'm still waiting to be approved by an administrator

I'll go through these later today. But if you give me your forum account name i'll register it once read.

albaldi, 2017-05-14:
> I'll go through these later today. But if you give me your forum account name i'll register it once read.

screwdriver
thanks

Fourdee, 2017-05-14:
@albaldi 

> screwdriver

Thanks, account now active.

albaldi, 2017-05-14:
> dietpi-config > Advanced > Time sync mode > Daemon + Drift

fixed my problem.

```
# timedatectl status
      Local time: Sun 2017-05-14 13:11:32 UTC
  Universal time: Sun 2017-05-14 13:11:32 UTC
        RTC time: n/a
       Time zone: Etc/UTC (UTC, +0000)
     NTP enabled: yes
NTP synchronized: yes
 RTC in local TZ: no
      DST active: n/a

```

Any idea why ntpd didn't work?

closed 2017-05-14 by albaldi

Fourdee, 2017-05-14:
@albaldi 

>  Any idea why ntpd didn't work?
6 Feb 15:22:35 ntpd[526]: Listening on routing socket on fd #22 for interface updates
 6 Feb 15:40:31 ntpd[526]: Deleting interface #5 eth0, fe80::ba27:ebff:fe5a:f195#123, interface stats: received=0, sent=0, dropped=0, active_time=1070 secs

Not sure, but appears it simply failed.

We may look at removing NTPD from DietPi, it is an outdated system. But we do love it for the ability to exit NTPD when its updated (saving resources).