MichaIng/DietPi #952: Not able to launch Kodi from LXDE

Invictaz, 2017-05-13:
System: Pi Zero
v149 running

Clicking on the kodi.desktop icon opens an empty leafpad. 
Kodi is not in the start menu. Kodi is installed AFTER LXDE was installed.

When you type Startkodi over SSH it runs but considerable slower than dedicated OpenElec 8.0 microSD. 

Exiting is not possible, exits to a black screen.

Fourdee, 2017-05-14:
@Invictaz 

Thanks 👍 

I'll take a look.

Fourdee, 2017-05-14:
- 🈯️ RPi 3 : Fine, Kodi 17.1
- RPi Zero:

Fourdee, 2017-05-14:
@Invictaz 

- Have you made any custom changes to your system, outside of the DietPi programs?
- Can you list installed software on this system ```dietpi-software list | grep ' =2 '``

Invictaz, 2017-05-14:
@Fourdee will reply with full log. Maybe uninstall and install Kodi to solve this?

Invictaz, 2017-05-14:
[image]

Invictaz, 2017-05-14:
Kernel

[image]

Invictaz, 2017-05-14:
@Fourdee 

Kodi 15.2 Git:02e7013 Compiled Dec 28 2015

Started with Startkodi & as window 2359

Error log attached 

[kodi_crashlog-20170514_213356.txt](https://github.com/Fourdee/DietPi/files/999672/kodi_crashlog-20170514_213356.txt)

Invictaz, 2017-05-14:
Screenshot to see that it is not opening via LXDE. And not in Sound/Video menu

[image]

Invictaz, 2017-05-14:
@Fourdee 

The issue I have is presented here. Strange is that it already suggests to start Kodi via Startkodi. But maybe my script is too old. Either way, upgrading to Kodi 17 sounds okay BUT Kodi 17 eats up more resources. Kodi 15.2 is fine for me.

https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=109088&p=750252#p750252

Fourdee, 2017-05-15:
@Invictaz 

DietPi-Software install of Kodi is via Raspbian apt repo. Current version is 17.1, previously it was 16.1. Not sure where/how 15.2 was installed on your system.

Maybe you have added additional apt repos that are conflicting with the install?:
```
ls -lha /etc/apt/sources.list.d
cat /etc/apt/sources.list
```

LXDE looks broken. I'd try a uninstall/reinstall of LXDE and Kodi. Failing that, please try a fresh install without any manual changes to system.

Invictaz, 2017-05-15:
@Fourdee All is installed via Dietpi-software. As said this SD card is already running since Dietpi 120 ish.

Invictaz, 2017-05-23:
[image]

Invictaz, 2017-05-23:
Setting is already at KODI setting 128mb GPU / 384MB RAM

Invictaz, 2017-05-23:
This problem is quite big and maybe not present in newer Kodi's. But I prefer to stay on 15. 

http://forum.kodi.tv/showthread.php?tid=268099
https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=91111
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=131742
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=139758
https://www.raspberrypi.org/forums/viewtopic.php?f=35&t=17473
http://forum.kodi.tv/showthread.php?tid=294884
https://www.reddit.com/r/raspberry_pi/comments/4i9b9p/kodi_black_screen_on_exit_raspbian_install/
https://gist.github.com/aperezm-vlex/75d7481afff0db8f5894

Invictaz, 2017-05-24:
@Fourdee 

The problem is partially fixed by reinstalling LXDE. 
Also the DietPi-Config / DietPi-Launcher and DietPi-Software shortcuts now work :+1: 

What is not fixed is that I cannot start Kodi from desktop. The shortcut still gives an empty file. 

Both kodi-standalone and startkodi work.

So if you could fix me a shortcut instead of launching it from terminal would be great.

closed 2017-05-25 by Invictaz