MichaIng/DietPi #953: (NanoPC-T3)Can not boot with-out SDCard.

nickfox-taterli, 2017-05-14:
NOTE:I can not post bugreport from my SBC.ftp timeout.

----------------------------------------------------------------
 DietPi : Bug Report
 14-05-2017
----------------------------------------------------------------
 Please fill out the information regarding your issue.

 When completed, save and exit nano:
 - Press CTRL+X
 - Press Y
 - Press Enter

----------------------------------------------------------------
 Your contact information
----------------------------------------------------------------
Email address = admin@lijingquan.net
Name          = TaterLi

----------------------------------------------------------------
 Short Description of the issue (eg: Desktop wont start):
----------------------------------------------------------------
Can not boot with-out SDCard.

----------------------------------------------------------------
 When does the issue occur? (eg: I tried to run startx):
----------------------------------------------------------------
Follow the instructions here to install DietPi to EMMC:
https://blog.stefanproell.at/2017/01/13/flashing-a-nanopc-t3-with-dietpi/

But there was a problem at startup from EMMC.

1)If I hold "boot" button and use SDCard(DietPi inside),boot from SDCard.
2)If I hold "boot" button and use SDCard(DietPi [NOT] inside),[NOT] booting.
3)If I does [NOT] hold "boot" button and use SDCard(DietPi inside),boot from EMMC.
4)If I does [NOT] hold "boot" button and use SDCard(DietPi [NOT] inside),[NOT] booting.
5)If I does [NOT] hold "boot" button,[NOT] booting.

----------------------------------------------------------------
 Error Codes (if applicable):
----------------------------------------------------------------
None

----------------------------------------------------------------
 Additional Notes:
----------------------------------------------------------------
None

End of Line ;)

nickfox-taterli, 2017-05-14:
adding: command_output_2.txt (deflated 73%)
  adding: command_output_3.txt (deflated 57%)
  adding: command_output_4.txt (deflated 52%)
  adding: command_output_5.txt (deflated 56%)
  adding: command_output_6.txt (deflated 84%)
  adding: command_output_7.txt (deflated 38%)
  adding: command_output_8.txt (deflated 71%)
  adding: command_output_9.txt (deflated 53%)
  adding: command_output_10.txt (deflated 75%)
  adding: command_output_11.txt (deflated 37%)
  adding: command_output_12.txt (deflated 63%)
  adding: command_output_13.txt (deflated 69%)
 [Info] Testing connection to dietpi.com
 [Info] Max duration of 20 seconds, please wait...
 [Ok] Connection test | Completed

 [Info] Uploading bug report, please wait...
--12:24:25-- `bugreport-609a9223-d799-4e59-b79a-579465e1f886-2.zip'
    => ftp://dietpi-survey:xxxxx@185.101.92.145:21/bugreport-609a9223-d799-4e59-b79a-579465e1f886-2.zip
Connecting to 185.101.92.145:21... connected!
Logging in as dietpi-survey ... Logged in!
Length: 15,119,558
 7% [==============>                                                                                                                                                                                  ] 1,170,432        160.7K/s ETA  1:24mError: Error encountered during uploading data (Success)
Send Failed (Abort successful) Waiting 2 seconds... Send Failed (UNIX Type: L8) FINISHED --12:24:40--
Transmission of 1 file failed.

ywjheart, 2018-02-12:
I have solved it, I can boot T3 from emmc, using dietpi v6.1.
https://ywjheart.wordpress.com/2018/02/10/write-image-to-nanopc-t3-emmc/

ywjheart, 2018-02-12:
───────────────────────────────────────
 DietPi     | 05:48 | Mon 12/02/18 
 ───────────────────────────────────────
 v6.1       | NanoPi M3/T3 (armv7l)
 ───────────────────────────────────────
 IP eth0    | 192.168.249.237
 ───────────────────────────────────────

 Created by : Daniel Knight
 Web        : http://DietPi.com
 Twitter    : http://twitter.com/dietpi_
 Donate     : http://goo.gl/pzISt9 
 DietPi's web hosting is powered by: MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

root@DietPi:~# fdisk -l
Disk /dev/mmcblk0: 7.3 GiB, 7818182656 bytes, 15269888 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x1bf0d6e1

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        2048   133119   131072   64M 83 Linux
/dev/mmcblk0p2      133120 15269887 15136768  7.2G 83 Linux

Disk /dev/mmcblk0boot1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Disk /dev/mmcblk0boot0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Fourdee, 2018-04-05:
Marking as resolved:
https://github.com/Fourdee/DietPi/issues/953#issuecomment-364833858

closed 2018-04-05 by Fourdee