MichaIng/DietPi #954: General | Default to SystemD-Timesyncd?

Fourdee, 2017-05-14:
investigate, remove NTPD? which is outdated.

albaldi, 2017-05-14:
because dietpi use
`ntpd -gq`
and because according to the man
```
-q
Exit the ntpd just after the first time the clock is set. This behavior mimics that 
of the ntpdate program, which is to be retired. The -g and -x options can be used 
with this option. Note: The kernel time discipline is disabled with this option.
```
one can simply use a cronjob with
> ntpdate -u

BTW on my raspy 1 model B (256MB ram)
```
ps -p 310 -o %cpu,%mem,vsize,rssize,cmd
%CPU %MEM    VSZ   RSS CMD
 0.0  0.8  12612  2020 /lib/systemd/systemd-timesyncd
```

Fourdee, 2017-05-29:
NTP synced:
```
timedatectl | grep -ci -m1 'NTP synchronized: yes'
```

Fourdee, 2017-05-31:
systemd-timesyncd is far too slow to verify sync (60's <>). We'll stick with NTP for now (10's <>).

closed 2017-05-31 by Fourdee