MichaIng/DietPi #955: 'rxvt-unicode-256color': unknown terminal type.

albaldi, 2017-05-14:
I'm using rxvt-unicode-256color on my PC and each time I ssh to my raspi I get this error
`'rxvt-unicode-256color': unknown terminal type.`

This error fordbid me to use `htop` or `nano`
Solution are export TERM to some compatible terminale
`export TERM='xterm-256color'`
Or copy from my pc to the raspi a file:
`scp /usr/share/terminfo/r/rxvt-unicode-256color user@server.com:.terminfo/r`

or simply install [`ncurses-term`](https://packages.debian.org/jessie/ncurses-term)

Reference:
http://alemani.com/urxvt-unknown-terminal-type-with-ssh/
https://raspberrypi.stackexchange.com/questions/53364/trying-to-ssh-in-using-urxvt-gives-error

Fourdee, 2017-05-14:
@albaldi 

Thanks, which SSH client are you running?

albaldi, 2017-05-14:
openssh 7.4p1-2
Reading the man I found out the ssh client can send to the server 
```
SendEnv
             Specifies what variables from the local environ(7) should be sent
             to the server.  Note that environment passing is only supported
             for protocol 2.  The server must also support it, and the server
             must be configured to accept these environment variables.  Refer
             to AcceptEnv in sshd_config(5) for how to configure the server.
             Variables are specified by name, which may contain wildcard char-
             acters.  Multiple environment variables may be separated by
             whitespace or spread across multiple SendEnv directives.  The
             default is not to send any environment variables.
```
but dropbear ssh doesn't support _AcceptEnv_

BTW it's not a problem, just wanted to have results when googling for "dietpi 'rxvt-unicode-256color': unknown terminal type"

thanks

Fourdee, 2018-04-05:
Marking as closed due to inactivity. Please reopen if required.

closed 2018-04-05 by Fourdee