MichaIng/DietPi #956: Swapfile is too big for my SD-card (768mb) v149

S60Team, 2017-05-14:
I have a problem on my RPI v1 (256mb RAM, 2Gb MicroSD):
Swapfile is too big for my SD-card (768mb) (v149)
Before that, i have ~50mb on v148 and it worked well.
Can you make a configuration for swapfile size in 150?
Please!

nickfox-taterli, 2017-05-15:
2GB MicroSD are too small....

S60Team, 2017-05-15:
But system is hang up...
So i copied all configs and reinstalled the system,
then changed swapfile in dietpi-config.

Fourdee, 2017-05-15:
@S60Team 

As of v149 (and the v148 RPI image), swapfile size is now 1GB - RAM. So in your case, 768MB~ would be generated for swapfile to prevent out of memory issues.

You can disable swapfile generation, after writing DietPi image, and, prior to booting device for 1st time in ```/boot/dietpi.txt```:
```Swapfile_Size=0```

closed 2017-05-17 by S60Team