MichaIng/DietPi #960: v150 breaks vmware image ?

sosher, 2017-05-15:
Hi, Thanks for dietpi 
my diet pi has been upgraded to version 150 on my vmware workstation , And since then 
I get the following error no matter which script I am running
"this script does not support the virtual machine image"

Is it something I am doing wrong ?
Is it possible to downgrade the version to 145 which worked ?

Thanks

powerzumsel, 2017-05-16:
dito
same issue to me
no "dietpi-xxxx" working

Fourdee, 2017-05-17:
@sosher 
@powerzumsel 

Thanks for the report 👍 

I'll take a look.

Fourdee, 2017-05-17:
Confirmed on fresh install > v150

Fourdee, 2017-05-17:
RootFS RW func check is the cause, script is terminated/failed due to VM script exclusion:
```
 [Info] Patching 146 to 147

This script does not support the Virtual Machine image.

/DietPi/dietpi/dietpi-drive_manager: line 41: Destroy: command not found
```

Fourdee, 2017-05-17:
This will be prevented and resolved in v151 release.

@sosher
@powerzumsel

#### If you experience this issue, please run the following commands to resolve:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-drive_manager -O /DietPi/dietpi/dietpi-drive_manager
sed -i 's#"044766be-96ca-491a-a93f-913b75a843bd"#044766be-96ca-491a-a93f-913b75a843bd#g' /etc/fstab
systemctl daemon-reload
```

sosher, 2017-05-17:
Thank you so much for the quick response and suggested fix.
I will be able to try the fix later today 
Thanks again !!!

powerzumsel, 2017-05-17:
fix is working
thank you very much !!!

Fourdee, 2017-05-17:
@sosher 
@powerzumsel 
Thanks guys 👍 

Marking as completed.

I'll try and get v151 pushed out within the next few days, to prevent this occurring during v150 update.

closed 2017-05-17 by Fourdee