MichaIng/DietPi #961: keyboard-settings not working and other problems with locale settings

DJCrashdummy, 2017-05-16:
when using `DietPi-Config` -> `Language/Regional Options` -> `Keyboard` it only shows the line `update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults` twice and immediately goes back to the previous screen where you can choose between _Local_, _Timezone_ and _Keyboard_...

---

additional i'm not sure if it is a bug, but at least it is wired:
i've set a different locale and when checking with the command `locale`, `LANGUAGE=` and `LC_ALL=` are empty, all other points are set correct.

_btw: if anyone is familiar with the locale-settings and can enlighten me...
why is every value set between double quotes (e.g. `LC_TIME="en_US.UTF-8"`) except `LANG=en_US.UTF-8`?_

Fourdee, 2017-05-17:
@DJCrashdummy 

> when using DietPi-Config -> Language/Regional Options -> Keyboard it only shows the line update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults twice and immediately goes back to the previous screen where you can choose between Local, Timezone and Keyboard...

A physical keyboard must be plugged into the device, for this to work, as the command ```dpkg-reconfigure keyboard-configuration``` appears to simply exit without one attached.

> btw: if anyone is familiar with the locale-settings and can enlighten me...
why is every value set between double quotes (e.g. LC_TIME="en_US.UTF-8") except LANG=en_US.UTF-8?

Quotations do not appear to be required. 
The reason for non-quoted entries is due to the way we also pump selected locale into ```/etc/environment``` (In an attempt to resolve: https://github.com/Fourdee/DietPi/issues/825):
```
cat << _EOF_ > /etc/environment
LC_ALL=en_GB.UTF-8
LANG=en_GB.UTF-8
_EOF_
```

Fourdee, 2017-05-24:
Marking as resolved. Please reopen if required.

closed 2017-05-24 by Fourdee

DJCrashdummy, 2017-06-04:
_sorry for my late reply, i was really busy in the last weeks._

> A physical keyboard must be plugged into the device, for this to work, as the command `dpkg-reconfigure keyboard-configuration` appears to simply exit without one attached.

oh sorry for bothering with this, you are right! but...

- a short message like **"no keyboard attached"** instead of just going back will prevent misunderstandings in the future; maybe some additional explanation like **"a specific layout can only be selected when a physical keyboard is attached."** will clarify everything.

- or at least a better description for the menu item(s) would be fine:
the language also for the keyboard is set with `Locale: Change Language and Regional Settings` but since only a specific layout is chosen with **`Keyboard`**, i would rather describe it **`Change Keyboard Layout (Setting)`** than `Change Keyboard Language Setting`!

> Quotations do not appear to be required.

@Fourdee: well... thanks for the info and the further enlightenment.

by the way: since you are obviously a pro, can you also tell me why `LANGUAGE=` and `LC_ALL=` are empty, when i check them with the command `locale`? - i noticed this also at other distributions...

DJCrashdummy, 2017-06-04:
open

Fourdee, 2017-06-04:
@DJCrashdummy 

> by the way: since you are obviously a pro, can you also tell me why LANGUAGE= and LC_ALL= are empty, when i check them with the command locale? - i noticed this also at other distributions...

Nobody is a pro in Linux 😉. Its impossible to know everything, yet alone keep up with its evolution.

As far as I know, ```LANGUAGE``` = isn't used. ```LC_ALL=``` should not be empty, its the core language control variable.

This is a fresh install on Odroid C2, seems fine:
```
root@DietPi:~# locale
LANG=en_GB.UTF-8
LANGUAGE=
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC="en_GB.UTF-8"
LC_TIME="en_GB.UTF-8"
LC_COLLATE="en_GB.UTF-8"
LC_MONETARY="en_GB.UTF-8"
LC_MESSAGES="en_GB.UTF-8"
LC_PAPER="en_GB.UTF-8"
LC_NAME="en_GB.UTF-8"
LC_ADDRESS="en_GB.UTF-8"
LC_TELEPHONE="en_GB.UTF-8"
LC_MEASUREMENT="en_GB.UTF-8"
LC_IDENTIFICATION="en_GB.UTF-8"
LC_ALL=en_GB.UTF-8
```

Can you double check the result (paste if you can)?
And also what was generated last time round:
```
cat /etc/locale.gen | grep -v '^#'
```

DJCrashdummy, 2017-06-04:
> Yep, good suggestion :+1: , i'll add a message box to inform users.

in my point of view instead/additional a more accurate description for **`Keyboard`** like **`Change Keyboard LAYOUT Setting`** would be fine, since `Change Keyboard LANGUAGE Setting` could be mistakable or is even false, or am i wrong?

---

```
root@DietPi:~# cat /etc/locale.gen | grep -v '^#'

de_DE.UTF-8 UTF-8
en_GB.UTF-8 UTF-8
```

> Can you double check the result (paste if you can)?
```
root@DietPi:~# locale
LANG=de_DE.UTF-8
LANGUAGE=
LC_CTYPE="de_DE.UTF-8"
LC_NUMERIC="de_DE.UTF-8"
LC_TIME="de_DE.UTF-8"
LC_COLLATE="de_DE.UTF-8"
LC_MONETARY="de_DE.UTF-8"
LC_MESSAGES="de_DE.UTF-8"
LC_PAPER="de_DE.UTF-8"
LC_NAME="de_DE.UTF-8"
LC_ADDRESS="de_DE.UTF-8"
LC_TELEPHONE="de_DE.UTF-8"
LC_MEASUREMENT="de_DE.UTF-8"
LC_IDENTIFICATION="de_DE.UTF-8"
LC_ALL=
```
and also after changing back to en_GB:

```
root@DietPi:~# locale
LANG=en_GB.UTF-8
LANGUAGE=
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC="en_GB.UTF-8"
LC_TIME="en_GB.UTF-8"
LC_COLLATE="en_GB.UTF-8"
LC_MONETARY="en_GB.UTF-8"
LC_MESSAGES="en_GB.UTF-8"
LC_PAPER="en_GB.UTF-8"
LC_NAME="en_GB.UTF-8"
LC_ADDRESS="en_GB.UTF-8"
LC_TELEPHONE="en_GB.UTF-8"
LC_MEASUREMENT="en_GB.UTF-8"
LC_IDENTIFICATION="en_GB.UTF-8"
LC_ALL=
```

although this is interesting, i doubt it is a bug (caused by DietPi) because i also noticed this on some other Debian-based distributions.

Fourdee, 2017-06-06:
@DJCrashdummy 

> Change Keyboard LAYOUT Setting

👍 Makes more sense. I'll get that updated for next commit.

Fourdee, 2017-06-06:
> although this is interesting, i doubt it is a bug (caused by DietPi) because i also noticed this on some other Debian-based distributions.

Strange. We also pump this into ```/etc/environment``` to ensure we cover all bases, can you verify ```LC_ALL= LANG=```?
```
cat /etc/environment
```

DJCrashdummy, 2017-06-08:
> Strange. We also pump this into `/etc/environment` to ensure we cover all bases, can you verify `LC_ALL= LANG=`?

```
root@DietPi:~# cat /etc/environment
LC_ALL=de_DE.UTF-8
LANG=de_DE.UTF-8
```
ok, here it is...
but just for completeness: on some other debian-based systems with empty `LC_ALL=` and `LANG=` i get with `cat /etc/environment` just the `PATH="..."`-variable.