MichaIng/DietPi #963: XU4 cpu frequency scaling (max frequency) - kernel 4.9

rncwnd79, 2017-05-16:
This belongs to kernel 4.9

The Samsung Exynos octacore has its little cores on the lower numbers: cpu0...cpu3 and its big cores on cpu4...cpu7.
Max freq for little cores is 1.4 GHz while maxfreq of the big cores is 2.0 GHz (1.8 GHz for the new passive cooled XU4Q).

When using a scaling capable governor like ondemand governor, dietpi-config uses /sys/devices/system/cpu/**cpu0**/cpufreq/scaling_max_/cpufreq/scaling_max_freq to gather the possible max frequency,
and cat /sys/devices/system/cpu/**cpu0**/cpufreq/scaling_available_frequencies to gather a list of possible frequencies.

That results in two problems:
1. Maybe this is a hardkernel kernel limitation, but the big cores run at 1.8 GHz by default.
2. dietpi-config / performance options / CPU Max Freq Limit is limited to 1.4 GHz and setting max freq to 2.0 GHz is not possible.

`root@odroid:/DietPi/dietpi# cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq`
`1400000`
`root@odroid:/DietPi/dietpi# cat /sys/devices/system/cpu/cpu4/cpufreq/scaling_max_freq`
`2000000`
`root@odroid:/DietPi/dietpi# cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies`
`200000 300000 400000 500000 600000 700000 800000 900000 1000000 1200000 1300000 1400000`
`root@odroid:/DietPi/dietpi# cat /sys/devices/system/cpu/cpu4/cpufreq/scaling_available_frequencies`
`200000 300000 400000 500000 600000 700000 800000 900000 1000000 1100000 1200000 1300000 1400000 1500000 1600000 1700000 1800000 1900000 2000000`

Fourdee, 2017-05-17:
@rncwnd79 

Thanks for the report 👍 

Yep, confirmed. I'll take a look, see if we can find the highest clock entry for all cores, then scale against that value.

Fourdee, 2017-05-17:
> Maybe this is a hardkernel kernel limitation, but the big cores run at 1.8 GHz by default.

With ondemand on my XU4 and 4.9, 1.3GHz / 2GHz:
```
root@DietPi-XU4:~# cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     armv7l
 Temp         |     51'c | Running warm, but safe.
 Governor     |     ondemand
 Throttle up  |     50% CPU usage

                 Current Freq    Min Freq   Max Freq
 CPU0         |      1300 MHz      200 MHz    1300 MHz
 CPU1         |      1300 MHz      200 MHz    1300 MHz
 CPU2         |      1300 MHz      200 MHz    1300 MHz
 CPU3         |      1300 MHz      200 MHz    1300 MHz
 CPU4         |      2000 MHz      200 MHz    2000 MHz
 CPU5         |      2000 MHz      200 MHz    2000 MHz
 CPU6         |      2000 MHz      200 MHz    2000 MHz
 CPU7         |      2000 MHz      200 MHz    2000 MHz
```

Fourdee, 2017-05-17:
For ticket reference:
https://github.com/Fourdee/DietPi/issues/926

rncwnd79, 2017-05-17:
Thanks for checking. I think I changed the max freq for the big cores once for heat reasons, just thought about it...

Fourdee, 2017-05-19:
Fixed for v151 with commit: https://github.com/Fourdee/DietPi/commit/5951116243e80434b99453b87273c68833cd9307

closed 2017-05-19 by Fourdee