MichaIng/DietPi #964: RetroPie missing and also unable to install manually

DJCrashdummy, 2017-05-16:
first it is really nice to have the option within `DietPi-Config` at `AutoStart Options` to choose `3 RetroPie`, so i started to look for RetroPie in `DietPi-Software` to install it as easy as all the other software (**thank you by the way!**), but sadly it wasn't there.
it would be very nice if RetroPie could also be installed via `DietPi-Software`!

---

but, no big deal... i went to [install RetroPie manually](https://retropie.org.uk/docs/Manual-Installation/), but this didn't work out either, because the installation stops with the message `Could not install package(s): libudev-dev libxkbcommon-dev libusb-1.0-0-dev libraspberrypi-dev.` and the only way to go further is `<OK>`.

so i had a look at the logs and i'm pretty sure the problem is the following:
```
The following packages have unmet dependencies:
 libraspberrypi-dev : Depends: libraspberrypi0 (= 1.20170427-1) but 1.20161215-1 is to be installed
E: Unable to correct problems, you have held broken packages.
Could not install package(s): libudev-dev libxkbcommon-dev libusb-1.0-0-dev libraspberrypi-dev.
```
so why are `libraspberrypi-bin`, `libraspberrypi0`, `raspberrypi-bootloader` and `raspberrypi-kernel` put on hold?

sfb2, 2017-05-16:
sorry, i´m not the "master" here ;-), but libraspberrypi-bin, libraspberrypi0, raspberrypi-bootloader and raspberrypi-kernel were "patched" here with a 384khz audio kernel for example and some other things..
So, they´d been put on hold to avoid being overwritten by the "standard" kernel files..

DJCrashdummy, 2017-05-16:
@sfb2: thanks for the info so far!

Fourdee, 2017-05-17:
@sfb2 

> So, they´d been put on hold to avoid being overwritten by the "standard" kernel files..

Spot on 👍 

> it would be very nice if RetroPie could also be installed via DietPi-Software!

@DJCrashdummy 

We originally had RetroPie in DietPi, very early on (sub v100 I believe). However, as RetroPie grew, it was impossible for us to fully control everything the RetroPie installer applies/installs. Let alone, predict what the user will install (binary/source build/emulators etc).

Therefor, we can't support it in **official capacity** and it was removed in ```dietpi-software```. However, it should run well on any DietPi system (after the unhold of kernel, assuming required by RetroPie).

Fourdee, 2017-05-17:
I've updated original post with the info/resolution:
https://github.com/Fourdee/DietPi/issues/964#issue-229005747

I'll mark this as known issue + workaround available.

k-plan, 2017-05-17:
@Fourdee 

you can do what ever you like, it will be a failure for someone.

- once someone cry to [get a newer kernel
](https://github.com/Fourdee/DietPi/issues/936)
-once some cry why software xyz is not more working after a auto kernel update

- once someone cry because the kernel is set on hold

- once some cry to have a 384khz audio support

Really sad to read such things. Better to to nothing ...

Fourdee, 2017-05-18:
@k-plan 

> you can do what ever you like, it will be a failure for someone.

Yep, cant win them all.

"Welcome to Linux" lol

I'll mark this as closed, initial question has been answered with fix option.

closed 2017-05-18 by Fourdee

Invictaz, 2017-05-18:
@k-plan I posted that question. It's not crying for a new kernel. It's just staying up to date. And for the letter of the fact, 4.9 isn't even the newest. I'm happy that it's done now.

If you want something support but newer kernels break it it's upto you to don't update. With newer features and bug fixes there is no such thing as no updating. Kernels are a tremendous amount of work and involve thorough testing. It's not that I was whining over a small software thingy not working.

DJCrashdummy, 2017-05-29:
_sorry for my late reply, i was really busy in the last weeks._

@Fourdee 
> We originally had RetroPie in DietPi, very early on (sub v100 I believe). However, as RetroPie grew, it was impossible for us to fully control everything the RetroPie installer applies/installs. ... Therefor, we can't support it in official capacity and it was removed in `dietpi-software`.

thanks for the detailed info! i just started using DietPi months ago, so i had no clue that RetroPie was already in `dietpi-software` and got removed...

@k-plan 
> Really sad to read such things.

from my side it was just a question, because it was not obvious to me why they are on hold. (and the "feature-request" is rejected for a good reason.)
if i bothered you with my interest in this project i'm sorry...