MichaIng/DietPi #966: [Request] adding Sonarr and Radarr to DietPi

Gravesy83, 2017-05-16:
In reference to the request at #339 , and in light of support and development on Couchpotato apparently going quiet recently (as i mentioned here; http://dietpi.com/phpbb/viewtopic.php?f=11&t=1882 ), i'd like to ask if there's any option to have Sonarr and Radarr added.
They have native support for a lot of Linux based systems (including the Raspberry Pi's), so they should, in theory, make a good addition to the list.

FireHelmet, 2017-05-21:
@Gravesy83 @Fourdee 

Installed and working well on odroid c2. Dan, I will send you an email about install procedure after few days of testing...

Thanks.

Fourdee, 2017-05-22:
@FireHelmet 

> Dan, I will send you an email about install procedure after few days of testing...

Legend 👍

FireHelmet, 2017-06-06:
@Gravesy83, scripts sent to @Fourdee

Gravesy83, 2017-06-06:
Now this is a pleasant surprise after a boring vacation :-)
Many thanks for doing this.

Fourdee, 2017-06-07:
Notes:
[image]

Fourdee, 2017-06-07:
#### Radarr Notes:
- 🈯️ Online Doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=7457#p7457

FireHelmet, 2017-06-07:
@Fourdee , yes Health messages are normal. You need to add Download client and indexer in sonarr config.

For download client, you can use Nzbd client or p2p client like torrent. And for indexer you can use the built-in indexer or prefer Jackett.

Fourdee, 2017-06-07:
@FireHelmet 

Great work on the install scripts, worked a charm 👍 :clinking_glasses: 

Both tested, completed.

closed 2017-06-07 by Fourdee