MichaIng/DietPi #967: Request V4L2 and HDMI Audio from 4.11 / 4.12

Invictaz, 2017-05-17:
https://github.com/torvalds/linux/commit/7b3ad5abf027b7643b38c4006d7f4ce47a86dd3a
https://www.phoronix.com/scan.php?page=news_item&px=Linux-4.11-Staging-Changes

http://www.phoronix.com/scan.php?page=news_item&px=RPi-VC4-HDMI-Audio-Patches

Fourdee, 2017-05-17:
@Invictaz 

4.12 is now available:
https://github.com/raspberrypi/linux/tree/rpi-4.12.y

When I get a chance, i'll compile and test it. If it seems stable, i'll offer it as alternative for testing.

Invictaz, 2017-05-17:
Legend!

Invictaz, 2017-07-05:
@Fourdee a final Linux 4.12 was released some days ago. Any chance it will be implemented?

Fourdee, 2018-02-22:
As we no longer provide custom kernels (we now use official RPi kernel via APT), i'll mark this as closed.

closed 2018-02-22 by Fourdee