MichaIng/DietPi #968: DietPi-Config | LED Control options

Fourdee, 2017-05-19:
eg: list all available (heartbeat/none etc) and apply.

Invictaz, 2017-05-19:
Would be great that a pattern is able to be configurable if full startup is achieved so I can run it headless without a screen. For example 3 blips.

Fourdee, 2017-05-19:
[image]
[image]

Fourdee, 2017-05-19:
@Invictaz 

> Would be great that a pattern is able to be configurable if full startup is achieved so I can run it headless without a screen. For example 3 blips.

The ```heartbeat``` option would benefit you for this. Basically flashes to a heartbeat once kernel is loaded (assumes booted)

Fourdee, 2017-05-20:
Worth noting that not all available trigger modes function as expected. This is dependent on device as we set trigger directly to ```/sys/class/leds/XXXX/trigger```

In general, the following work on most boards:
- heartbeat
- none (off)

Fourdee, 2017-05-20:
Completed.

Fourdee, 2017-05-27:
The order + amount of ```/sys/class/leds``` entries change when devices are attached (eg: keyboard). So we need to match names in the save file to prevent incorrect LED triggers being applied.

Fourdee, 2017-05-27:
Done.

closed 2017-05-27 by Fourdee