MichaIng/DietPi #969: OdroidXU4 recognized as RPi (v150)

filemoon, 2017-05-20:
Hi.

I have 3 Odroid XU4 boards (with cloudshell) after upgrade to v150 one is recognised as Rpi.
This is content of .hw_model file

0
RPi  (armv7l)
3
/dev/mmcblk0p2
79853dc4-02d9-45a1-bbae-208dae17dac9
2
armv7l

0
0
0100
Unknown
Unknown
0
Unknown

Fourdee, 2017-05-20:
@filemoon 

Thanks for the report 👍 

Is this running the 3.x kernel?
```uname -a```

filemoon, 2017-05-20:
I have very same result for all my XU4 boxes.

#1 SMP PREEMPT Sat Apr 8 00:28:48 CEST 2017 armv7l GNU/Linux

cat /etc/.dietpi_image_version (file is not present on any of my arm boxes)

processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 24.00
Features	: swp half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 24.00
Features	: swp half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 24.00
Features	: swp half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 24.00
Features	: swp half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 3

processor	: 4
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 12.00
Features	: swp half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x2
CPU part	: 0xc0f
CPU revision	: 3

processor	: 5
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 12.00
Features	: swp half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x2
CPU part	: 0xc0f
CPU revision	: 3

processor	: 6
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 12.00
Features	: swp half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x2
CPU part	: 0xc0f
CPU revision	: 3

processor	: 7
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 12.00
Features	: swp half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x2
CPU part	: 0xc0f
CPU revision	: 3

Hardware	: ODROID-XU3
Revision	: 0100
Serial		: 0000000000000000

PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Fourdee, 2017-05-20:
@filemoon 

> ID=raspbian

Strange, this is reporting Raspbian OS and being picked up by our script.

Just tried a fresh image installation of v145 image, results are fine:
```
ID=debian
```

I'am not entirely sure how your ```/etc/os-release``` contains ```ID=raspbian```, however, we can override this and force XU4 id with the following commands:
```
echo 11 > /etc/.dietpi_hw_model_identifier
/DietPi/dietpi/dietpi-obtain_hw_model
```

Then check valid with:
```
cat /DietPi/dietpi/.hw_model
```

filemoon, 2017-05-20:
Looks fine now.
I have no idea why systems thinks is Raspbian (there was old DietPi image use to build them).

Fourdee, 2017-05-24:
As unable to replicate on our current images, i'll mark this resolved.

closed 2017-05-24 by Fourdee