MichaIng/DietPi #970: RPi | Support for USB write > boot

Fourdee, 2017-05-21:
> http://dietpi.com/phpbb/viewtopic.php?f=9&t=1899#p7268

Image update required.

Fourdee, 2017-05-21:
PARTUUID's required:

- ```/boot/cmdline.txt``` ```root=PARTUUID=84fa8189-02```
- ```/etc/fstab``` 
```
PARTUUID=84fa8189-01  /boot           vfat    defaults          0       2
PARTUUID=84fa8189-02  /               ext4    defaults,noatime  0       1
```

Invictaz, 2017-05-21:
Will boot off USB be also supported on other platforms e.g. Pi Zero?

Fourdee, 2017-05-21:
@Invictaz 

> Will boot off USB be also supported on other platforms e.g. Pi Zero?

Should be fine. AFAIK this is a RPi feature only, built into kernel/uboot.

Invictaz, 2017-05-21:
@Fourdee Great. Looking forward to it.

Fourdee, 2017-05-22:
@Invictaz 

Possibly RPi 3 only:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

Invictaz, 2017-05-23:
@Fourdee 

Normally this goes through Uboot?

It's partly possible already. Some code is loaded off SD and some of USB

https://www.raspberrypi.org/forums/viewtopic.php?t=12015&f=66

https://raspberrypi.stackexchange.com/questions/62965/boot-zero-w-from-usb-device

Seems to involve a bootcode.bin

Fourdee, 2017-05-26:
> Starting with the 2017-04-10 release of Raspbian you can install a working Rasbian system to a USB mass storage device by copying the operating system image directly onto your USB device, in the same way that you would for an SD card.

Hmm, I cant get any of my USB mass storage devices to boot with latest Raspbian:
Raspbian Lite 2017-04-10:
- 🈴 No boot with SanDisk SDCZ43-032G-G46 Ultra Fit USB Flash Drive USB
- 🈴 No boot with Lexar JumpDrive S45 16 GB

Fourdee, 2017-05-26:
New RPi image live with USB boot support (still requires bit set prior https://github.com/Fourdee/DietPi/issues/970#issuecomment-304264851):
http://dietpi.com/downloads/images/DietPi_RPi-armv6-(Jessie).7z

k-plan, 2017-05-26:
@Fourdee 

have you missed [this](https://github.com/Fourdee/DietPi/issues/509#issuecomment-246734927)?

- Yes, USB boot without a SD Card only for RPi3.

- Yes, you have first to change OTP value to `17:3020000a` by  adding `program_usb_boot_mode=1` to `config.txt` and reboot.

- And OTP changes can not undo !! 
_(BTW: -if something goes wrong by changing OTP value, you loosing RPi functionality e.g. onboard WiFi -  sadly self tested!)_

-  not all flash drive will work

- problems with USB-HDD (spinning disk) timeout

- it will increase boot time about 5 -10 sec.

Yes, it's possible to speed it up by putting a empty FAT formatted SD card in the slot, but so I can as well stay on this, which will work on every RPi.
Yes, you can put only a `bootcode.bin` on Fat formated SD-Card and let it in your RPi3, so it will be able to boot from more usb devices without problems.
Yes, you can add  `program_usb_boot_timeout=1` into the `config.txt`. This improves to around 25% of successful start-ups, but no more than that. And it can speed-up the boot time a little bit.

You can test, if USB R/RW speed come up to a U3 sd card + overclocking the sd to 100 MHz

Fourdee, 2017-05-26:
@k-plan 

> And OTP changes can not undo !!
(BTW: -if something goes wrong by changing OTP value, you loosing RPi functionality e.g. onboard WiFi - sadly self tested!)

Wasn't aware of that! Thanks for the heads up.

I personally wont be using USB boot/rootfs due to USB + Eth bandwidth shared. But, at the very least we need to offer and support this "functionality" to our users with the updated image.

> You can test, if USB R/RW speed come up to a U3 sd card + overclocking the sd to 100 MHz

With my Sandisk ultra fit USB3.0, 40MB/s read, same as UHS3 and 100MHz overclock.

Fourdee, 2017-05-27:
Completed.

closed 2017-05-27 by Fourdee

k-plan, 2017-05-27:
@Fourdee 

Experience and testing results:

```
───────────────────────────────────────
 DietPi     | 16:59 | Sat 27/05/17
 ───────────────────────────────────────
 V150       | RPi 3 Model B (armv7l)
 ───────────────────────────────────────

```

```
root@RPi-3:~# vcgencmd otp_dump | grep 17:
17:1020000a
```
- SD-card: [SanDisk Extreme PLUS 16 GB microSDHC U3](https://www.amazon.de/SanDisk-Extreme-microSDHC-Speicherkarte-SD-Adapter/dp/B010NE3ND8)

[image]

------------------------------------

```
root@RPi-3_USB:~# vcgencmd otp_dump | grep 17:
17:3020000a
```

- USB flash drive: [SanDisk Ultra Fit 64 GB  USB 3.0 150 MB/Sec](https://www.amazon.de/SanDisk-Ultra-Fit-USB-Flash-Laufwerk-150-MB/dp/B01BGTG3JA)

[image]

-------------------------------------

- with a little bit of RPi 3 over-clocking and sd-card tweaking:

[image]

- SD-card: [SanDisk Extreme PLUS 16 GB microSDHC U3](https://www.amazon.de/SanDisk-Extreme-microSDHC-Speicherkarte-SD-Adapter/dp/B010NE3ND8)

[image]

- USB flash drive: [SanDisk Ultra Fit 64 GB  USB 3.0 150 MB/Sec](https://www.amazon.de/SanDisk-Ultra-Fit-USB-Flash-Laufwerk-150-MB/dp/B01BGTG3JA)

[image]

> If SD reader went through usb hub, dtoverlay=sdhost,overclock_50=100 will as well overclock usb host.
> 
> I don't think so, but you can test it.

Tested now, yes => _busted_ √ 😃 

> With my Sandisk ultra fit USB3.0, 40MB/s read, same as UHS3 and 100MHz overclock.

Sorry, not for me.

---------------------------------

As well trying to get RPi-3 Boot from Mass Storage Device (MSD) with two different SSD  and [USB 3.0 to SSD adapter](https://www.amazon.de/Sabrent-5-Zoll-SATA-Festplatten-Optimiert-Unterstützt-EC-SSHD/dp/B011M8YACM) 

No luck to get it on work ....

... not with `program_usb_boot_timeout=1`
... not with a FAT formated sd-card with `bootcode.bin` 👎 

----------------------------------

And yes, 

> [Be warned that this feature is experimental and does not work with all USB mass storage devices. 
Since 2017-04-10 release ..](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md)

> I personally wont be using USB boot/rootfs due to USB + Eth bandwidth shared.

Yes! Same here. SD card tweaking rocks! 😄   👍 

> But, at the very least we need to offer and support this "functionality" to our users with the updated image/sourcecode.

Anyway 😉 . 

Would be nice to but this into `dietpi-config` for the users.

You can read out `vcgencmd otp_dump | grep 17:` and if it is not set, put it in `/DietPi/config.txt` once and force a reboot like you do, when changing the hostname.
This would be nice for users!

Do you want a new issue for this❓ 

_BTW: `program_usb_boot_mode=1` will change same more values in `otp`:_

```
17:1020000a  =>  3020000a
18:1020000a  =>  3020000a
66:00000000  =>  00009daa
```

Fourdee, 2017-05-28:
@k-plan 

> Would be nice to but this into dietpi-config for the users.
You can read out vcgencmd otp_dump | grep 17: and if it is not set (disabled), user can easy change it to enable , so it will put in /DietPi/config.txt once and force a reboot like you do, when changing the hostname.
This would be nice for users!
Do you want a new issue for this

Yep, sounds good, create the ticket 👍

k-plan, 2017-05-28:
@Fourdee 

> Yep, sounds good, create the ticket 👍

Done, see:  https://github.com/Fourdee/DietPi/issues/983