MichaIng/DietPi #971: DietPi-Software | PlexPy

Fourdee, 2017-05-21:
> https://github.com/JonnyWong16/plexpy

FireHelmet, 2017-05-21:
@Fourdee 

Can you assign to me this job ?

Thanks

Fourdee, 2017-06-08:
Doesnt seem able to function with PMS server
```
There was an error communicating with your Plex Server.
```

Yet connections are confirmed:
[image]

Logs filled with errors, and PlexPy thinks its outdated:
[image]

Fourdee, 2017-06-08:
- 🈯️ Online Doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=7463#p7463

FireHelmet, 2017-06-08:
@Fourdee , what is the port used in the settings of plexpy ? And yes, plexpy need git to check version...sorry.

I will try the latest version of plexpy in my DietPi. I keep you informed.

Thanks

Fourdee, 2017-06-09:
> what is the port used in the settings of plexpy ?

@FireHelmet 

For web interface + settings? its :8181

Fourdee, 2017-06-12:
🈯️ Tested on RPi, completed.

closed 2017-06-12 by Fourdee